acado/acado #130: Installation is broken!

skaynama, 2015-01-12:
I think after the latest changes ACADO no longer compiles. The output of "make" is:

skaynama@nb-skaynama:~/ACADOtoolkit/build$ make
Scanning dependencies of target acado_casadi
[  0%] Building CXX object external_packages/casadi/CMakeFiles/acado_casadi.dir/symbolic/printable_object.cpp.o
[  0%] Building CXX object external_packages/casadi/CMakeFiles/acado_casadi.dir/symbolic/shared_object.cpp.o
Linking CXX static library ../../lib/libacado_casadi.a
[  0%] Built target acado_casadi
Scanning dependencies of target acado_csparse
[  0%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_tdfs.c.o
[  0%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_maxtrans.c.o
[  1%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_schol.c.o
[  1%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_counts.c.o
[  1%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_cholsol.c.o
[  1%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_lsolve.c.o
[  2%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_print.c.o
[  2%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_lu.c.o
[  2%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_etree.c.o
[  2%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_leaf.c.o
[  2%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_usolve.c.o
[  3%] Building C object external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_scatter.c.o
/tmp/ccWvVHyJ.s: Assembler messages:
/tmp/ccWvVHyJ.s:61: Error: no such instruction: `vfmadd312sd (%r9),%xmm0,%xmm1'
make[2]: *** [external_packages/csparse/CMakeFiles/acado_csparse.dir/cs_scatter.c.o] Error 1
make[1]: *** [external_packages/csparse/CMakeFiles/acado_csparse.dir/all] Error 2
make: *** [all] Error 2

rienq, 2015-01-12:
Hi there,

It is possible that you might run into problems with certain compilers as indeed some things have changed in the current stable version compared to the previous one, but it should not be too difficult to get it to work..
Which operating system and compiler versions are you using (both for C and C++)?

Also, could you try to remove the build folder completely after pulling the latest version to be sure that the problem is reproducible? 
I haven't seen such an error myself before.. although, I have seen quite some posts online where they propose to use a more up to date gcc/g++ compiler.

skaynama, 2015-01-12:
Thanks for the quick response!

I'm using Linux (Ubuntu 14.07) running on Intel Core i5 vPro. My version of gcc/g++ is 4.6.3.

I've already tried removing the build directory. I can confirm that the problem is reproducible.

rienq, 2015-01-12:
No worries and thanks for the extra info.. 4.6.3 is a few years old though, would it be possible to try out a newer version e.g. 4.7/4.8?

skaynama, 2015-01-15:
I primarily work with ROS Hydro, which unfortunately limits my ability to upgrade the OS and therefore my g++/gcc distribution. Is there anyway around this?

mvukov, 2015-01-15:
1) try to change the compiler to e.g. clang or newer gcc; you don't have to upgrade the OS, just a few packages or
2) https://github.com/acado/acado/blob/master/cmake/CompilerOptions.cmake#L96-L97 comment out those two lines.

skaynama, 2015-01-15:
Yep, that solved the problem! Thank you.
ACADO is a great package. Well done folks.

closed 2015-01-15 by mvukov