acado/acado #134: Code generation bug (bug/issue)

UPC-EspenSkjong, 2015-02-18:
Hi,

I've found a couple of problems with the code generation.
1. When using EXACT_HESSIAN the file "acado_hessian_regularization.c" is not added to the make tree, my work-around was to make a dummy header file "acado_hessian_regularization.h" and added to the build tree.
2. Compile error: acado_qpoases_interface.cpp, in function getErrorString(). The MessaheHandling does not have a getErrorString-function.
3. Trying to use OnlineData to provide changeable parameters for my generated code:
* Compile error when using default constructor "OnlineData data;" The OnlineData parameter is used as a regular parameter in some of the differential equations. The workaround is to call data(0,0).
* In the generated code the OnlineData parameter is not shown at all. ACADO_NOD = 0, and the acadoVariables struct does not include any online data.

mvukov, 2015-02-18:
2. Copy the newest version of qpOASES to your export folder; from `acado/external_packages/qposes`
3.2. I heard this recently. I think it's the problem with the new(er) versions of ACADO. Contact @rienq

UPC-EspenSkjong, 2015-02-18:
Do you know if this will be fixed in the near future, or if there is an older version of ACADO supporting EXACT_HESSIAN in code generation and supports OnlineData data type?

mvukov, 2015-02-18:
@rienq  can give you more info.

UPC-EspenSkjong, 2015-02-18:
Alright, the solution you proposed about the newest version of qpOASES did not pan out. It fixed the problem with MessageHandling, however, the solution from the QP is not correct. I tracked this issue and found out the "solve" procedure in "acado_qpoases_interface.cpp" is different. The usage of QProblemB gives the correct answer, but the QProblem (which is generated from the new qpoases folder) fails...

mvukov, 2015-02-18:
I assume now your issue (2) is OK.

What do you mean by "different" for the solve procedure?

When QProblem fails, what it means? Do you give all necessary info to it? QProblem is instantiated only when a QP formulation has affine constraints.

UPC-EspenSkjong, 2015-02-18:
I'm sorry for such a bad explanation. Once I updated the qpoases folder and recompiled everything, the auto-generated MPC did not solve the problem correctly, i.e. all controls were zero, and the differential states were not correct.  I noticed that the `acado_qpoases_interface.c` file used the `QProblem` class, which is different from the `QProblemB` class used before I updated the qpoases folder. I swapped the `QProblem` instance with a `QProblemB` instance, and everything worked, i.e. swapped 
```javascript
int solve( void )
{
	nWSR = QPOASES_NWSRMAX;

	QProblem qp(120, 1);
	
	returnValue retVal = qp.init(acadoWorkspace.H, acadoWorkspace.g, acadoWorkspace.A, acadoWorkspace.lb, acadoWorkspace.ub, acadoWorkspace.lbA, acadoWorkspace.ubA, nWSR, acadoWorkspace.y);

	qp.getPrimalSolution( acadoWorkspace.x );
	qp.getDualSolution( acadoWorkspace.y );
...
```
with
 
```javascript
int solve( void )
{
	nWSR = QPOASES_NWSRMAX;

	QProblemB qp( 120 );
	
	returnValue retVal = qp.init(acadoWorkspace.H, acadoWorkspace.g, acadoWorkspace.lb, acadoWorkspace.ub, nWSR);

	qp.getPrimalSolution( acadoWorkspace.x );
	qp.getDualSolution( acadoWorkspace.y );
...
```

When I recompile (generate new code) the `acado_qpoases_interface.c` is updated and uses the `QProblem` class. Unfortunately I'm not able to get the MPC working without swapping the `QProblem` instance with `QProblemB`, as listed above. I've tried to tune the MPC, but with no luck. However, the class swapping does the trick needed, but it would be nice to know if this is an issue or not for future references.

As you assumed, issue (2) is solved. The MPC has great real-time properties, thanks to you guys, and I hope that the issues regarding the OnlineData functionality will be solved, giving me the opportunity to actually embed the MPC in a real-life application.

mvukov, 2015-02-19:
OK, let's start from a beginning.

- I don't know what problem are you actually solving, but you should always penalize controls.
- Do you monitor return code from the `feedbackStep`? If it's not zero, there is an error. You can use `getErrorString` to find which one.
- If you switch to `QProblemB`, you won't satisfy your path/state constraints but only input constraints. It's equivalent to removal of state and path constraints in the OCP formulation. So, this indicates that the problem might be in your formulation and/or initial guess you give to the solver.

UPC-EspenSkjong, 2015-02-19:
I use the `minimizeLagrangeTerm` objective, so I do not have path constraints. I do not have state constraints either because all states are naturally bounded in the problem. I checked the return code from `feedbackStep` and only the first step returned successfully. So there is something wrong.

mvukov, 2015-02-19:
Hmm... Let's not mix apples and oranges. Specific objective minimization does not have anything in common with path constraints.

If you don't have path nor state constraints in your formulation, and you see `QProblem` (not `QProblemB`) in the exported interface to qpOASES that is certainly a bug. @rienq might have a look at this, since he is developing the exact Hessian solver.

UPC-EspenSkjong, 2015-02-19:
I think it is correct the way it is. I do not need the start and end constraint for my problem, and by removing those the `QProblemB` is instantiated by the code generation. I think the problem was with my formulation. Sorry for the inconvenience.

rienq, 2015-02-24:
Hi there,

I seem to be a bit late to join the discussion but a few short remarks/questions:
1) First, you should know that the Exact Hessian based solver is research work in progress at the moment and this certainly regarding the regularization where things should still improve in the (near) future. Then, the file "acado_hessian_regularization.c" is not included where exactly? I suppose you are using ACADO from C++ and not from MATLAB? 
I guess the C Makefile templates indeed still need to be updated, I can easily fix that..

3.1) What is the problem with the OnlineData constructor exactly?
3.2) Also here I am not sure what you are doing.. There is a known issue though when you are using OnlineData which does not appear in the model equations but instead only appears in the residual function for the objective for example. There is only a quick (and dirty) fix for this at the moment..

UPC-EspenSkjong, 2015-02-25:
Hi,
1) I'm, as you said, modeling in C++. The `acado_hessian_regularization.c` is missing its header file (?) in the generated code, and is not added to the make file build tree.
3.1) I guess the `OnlineData` object can be used to give online parameters to the exported code, e.g. such as amplitudes and phases for a sine wave? When trying to use the `OnlineData` object, in the same way all other objects in acado are used, e.g. `DifferentialState`, `Control`, `Parameter`, `IntermediateState`, it fails. Compile errors indicates that there is some wrong with the dimension of the `OnlineData` object instantiated. For example:
```

double omega = 2*PI*50;

OnlineData a;
OnlineData p;

DifferentialState s;
DifferentialState t;

DifferentialEquation f;

f << dot(t) == 1;
f << dot(s) == a*cos(omega*t + p); // Compile error: Fix -> a(0,0) and b(0,0)

```

The code above will not give any physical meaning, anyway, the code will not compile unless the OnlineData instances are called by `a(0,0)` and `b(0,0)`. In the exported code there is no indication that the OnlineData instances are present in the compiled code as `ACADO_NOD` equals zero, and I suspect the OnlineData object has no role in the exported code at all.

rienq, 2015-02-25:
Hey,

1) There is not supposed to be a corresponding header file, it's just that the "acado_hessian_regularization.c" file is not added to the Makefile yet (since this code was developed using the MATLAB interface, but this can be fixed easily).

3) You should not access the OnlineData objects like you do e.g. "a(0,0)", "b(0,0)".. I mean, obviously that is correct as well but there is no reason why that should be necessary. Also, your code seems to work just fine on my system here so if you could somehow provide us with your complete code we could maybe figure out what is going on but I assume there is something else going on in your code, you are surely not getting any error messages from ACADO?

UPC-EspenSkjong, 2015-02-25:
When running code-export (C++ executable) the following message is given:
```
NOTE: You are using the new (unstable) N2 condensing feature for exact Hessian based RTI..
[ACADO] Error: Incompatible vector dimensions
  Code: (RET_VECTOR_DIMENSION_MISMATCH) 
  File: /<usr-path>/acado/code_generation/export_arithmetic_statement.cpp
  Line: 561

[ACADO] Error: Unable to export statement due to incomplete definition
  Code: (RET_UNABLE_TO_EXPORT_STATEMENT) 
  File: /<usr-path>/acado/code_generation/export_statement_block.cpp
  Line: 282

```
The fix I proposed earlier fixes the issues, but `ACADO_NOD` in the exported code is zero. 

If you give me an e-mail addr. I'll send the implementation. You can respond to espen.skjong@ulstein.com

rienq, 2015-02-26:
Hi there,

Thank you for sending me your code.. could you try pulling the latest master and rerun your code?
By the way, one useful thing to know maybe (certainly when running code which is still under development like the exact hessian based RTI solver) is how to set the logger to "debug mode" using one line:

Logger::instance().setLogLevel( LVL_DEBUG );

Cheers

UPC-EspenSkjong, 2015-03-02:
Thanks! Seems to be working now :-)

rienq, 2015-03-02:
Great!

closed 2015-03-02 by rienq