acado/acado #136: New tagged release?

mikepurvis, 2015-03-02:
Looks like the last tagged version was in Dec 2013. Any chance we can get the current Stable branch tagged 1.2.1 or 1.2.2?

mikepurvis, 2015-03-13:
@mvukov Do you have a sense what the timing will be for this? If it's far enough off in the future, it might make more sense for me to just fork the repo and tag that, and then switch back to the upstream source repo tag when the next release is available. Thoughts?

mvukov, 2015-03-15:
At the moment I think you should follow your suggestion. Especially if your user code is bounded to a specific API version.

SimplyDanny, 2017-02-19:
After another two years, I think it's time to ask again for some regular tags. 😊 

As @mikepurvis mentioned, tags are usually necessary for ACADO being accepted in packaging systems. I already created a Formula for the [Homebrew](https://github.com/Homebrew/brew) package manager. Unfortunately, it won't be included without tagged versions.

In my opinion, installation with just one simple command is conducive to the popularity of every software project.

mvukov, 2017-02-19:
As far as I am aware, there is no agenda towards a next release of ACADO. If you just need a tag of the repo to be able to include in Homebrew, which is awesome of course, let me know and we can make a tag for you.

SimplyDanny, 2017-02-19:
Great! Some tags like the `*beta` ones on the `stable` branch from time to time are totally sufficient.

Thank you!

SimplyDanny, 2017-02-20:
The Formula for Homebrew is ready now. Please make a tag for me @mvukov . 😄

mvukov, 2017-02-22:
@SimplyDanny [Here](https://github.com/acado/acado/releases/tag/v1.2.2beta) is the release. Please let me know if you need additional support.

SimplyDanny, 2017-03-06:
ACADO can now be [installed](https://github.com/Homebrew/homebrew-science/pull/5183) with Homebrew. I updated the installation instructions accordingly. See [here](https://github.com/acado/acado.github.io/pull/2).

Thank you again for the tag!