acado/acado #139: Master broken?

mvukov, 2015-03-05:
Commit c7f1361 screwed up the master?

mvukov, 2015-03-05:
@ferreau ?

mvukov, 2015-03-05:
```
Conflicts:
	CMakeLists.txt
	acado/bindings/acado_qpoases/qp_solver_qpoases.hpp
	examples/CMakeLists.txt
	external_packages/qpOASES-3.1.0/CMakeLists.txt
	interfaces/matlab/shared/objects.m
```

ferreau, 2015-03-05:
Please revert this commit, I would not know how to do without creating even more mess. Sorry for that.

mvukov, 2015-03-05:
I am on vacation, @rienq might take care about this issue.

rienq, 2015-03-05:
I have no clue what happened here and the fix I found online did not do exactly what I hoped it would do in this case, but I have the impression the files are back to what they were before Joachim's commits.. I will leave this issue open for now though.

We also have to find a better way of pushing Joachim's fixes: @ferreau, I guess you should locally try to merge the acado master (instead of your own master) with your fixes and then push that merged version. I have the impression that your additions of the new versions of qpOASES were performed over a very old version of ACADO which created all the conflicts later on.

closed 2015-03-05 by rienq

ghorn, 2015-03-06:
This could be prevented in the future if people submit pull requests instead of just pushing to master. Travis-ci would have said "this pull request breaks things".