acado/acado #140: Error building pendulum_dae_nmpc_test in master branch

mkotlyar, 2015-03-06:
`make` outputs:

[ 63%] Generating ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_common.h, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_solver.c, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_integrator.c, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_qpoases_interface.hpp, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_qpoases_interface.cpp, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_auxiliary_functions.h, ../../../../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_auxiliary_functions.c

ACADO Toolkit::Code Generation Tool
Copyright (C) 2008-2014 by Boris Houska, Hans Joachim Ferreau,
Milan Vukov, Rien Quirynen, KU Leuven.
Developed within the Optimization in Engineering Center (OPTEC)
under supervision of Moritz Diehl. All rights reserved.

ACADO Toolkit is distributed under the terms of the GNU Lesser
General Public License 3 in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Lesser General Public License for more details.

Segmentation fault (core dumped)
examples/code_generation/mpc_mhe/CMakeFiles/pendulum_dae_nmpc_test.dir/build.make:56: recipe for target '../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_common.h' failed
make[3]: *** [../examples/code_generation/mpc_mhe/pendulum_dae_nmpc_export/acado_common.h] Error 139
CMakeFiles/Makefile2:6008: recipe for target 'examples/code_generation/mpc_mhe/CMakeFiles/pendulum_dae_nmpc_test.dir/all' failed
make[2]: *** [examples/code_generation/mpc_mhe/CMakeFiles/pendulum_dae_nmpc_test.dir/all] Error 2
CMakeFiles/Makefile2:6020: recipe for target 'examples/code_generation/mpc_mhe/CMakeFiles/pendulum_dae_nmpc_test.dir/rule' failed
make[1]: *** [examples/code_generation/mpc_mhe/CMakeFiles/pendulum_dae_nmpc_test.dir/rule] Error 2
Makefile:2061: recipe for target 'pendulum_dae_nmpc_test' failed
make: *** [pendulum_dae_nmpc_test] Error 2
```

mkotlyar, 2015-03-09:
gcc and g++ version 4.7.4

rienq, 2015-03-09:
this is weird, I have no problems on my system using gcc and g++ version 4.8.2 neither using Clang version 3.4.0.. did you have this segfault before as well with the same compiler? 
Because I cannot imagine what could have changed for this example recently..

mkotlyar, 2015-03-10:
It seems to be a compiler-related issue. With gcc/g++ 4.8.3 it does build.

ghorn, 2015-03-10:
@mkotlyar could you run the example in gdb and paste the backtrace?

mvukov, 2015-03-13:
@mkotlyar please run valgrind on this one and paste the output.

mkotlyar, 2015-03-16:
The command that segfaults is:
```
cd ACADOtoolkit/examples/code_generation/mpc_mhe
./code_generation_pendulum_dae_nmpc
```
Running from gdb:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff760c307 in Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const>, 3, 0>::run(Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const> const&, Eigen::internal::scalar_sum_op<double> const&) ()
   from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
(gdb) backtrace
#0  0x00007ffff760c307 in Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const>, 3, 0>::run(Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const> const&, Eigen::internal::scalar_sum_op<double> const&) ()
   from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#1  0x00007ffff760c5ba in Eigen::internal::isApprox_selector<Eigen::Matrix<double, -1, -1, 1, -1, -1>, Eigen::Matrix<double, -1, -1, 1, -1, -1>, false>::run(Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, double const&) () from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#2  0x00007ffff78b2cb4 in ACADO::ExportNLPSolver::setLSQObjective(ACADO::Objective const&) () from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#3  0x00007ffff78b5daa in ACADO::ExportNLPSolver::setObjective(ACADO::Objective const&) () from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#4  0x00007ffff788d5a3 in ACADO::OCPexport::setup() () from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#5  0x00007ffff788de30 in ACADO::OCPexport::exportCode(std::string const&, std::string const&, std::string const&, int) () from /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1
#6  0x0000000000406eee in main ()
```

mkotlyar, 2015-03-16:
@mvukov , valgrind output:
```
mkatliar@almond-vm-ubuntu:~/ACADOtoolkit/examples/code_generation/mpc_mhe$ valgrind ./code_generation_pendulum_dae_nmpc
==31318== Memcheck, a memory error detector
==31318== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==31318== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==31318== Command: ./code_generation_pendulum_dae_nmpc
==31318== 

ACADO Toolkit::Code Generation Tool
Copyright (C) 2008-2014 by Boris Houska, Hans Joachim Ferreau,
Milan Vukov, Rien Quirynen, KU Leuven.
Developed within the Optimization in Engineering Center (OPTEC)
under supervision of Moritz Diehl. All rights reserved.

ACADO Toolkit is distributed under the terms of the GNU Lesser
General Public License 3 in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Lesser General Public License for more details.

==31318== Use of uninitialised value of size 8
==31318==    at 0x4F842E3: Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const>, 3, 0>::run(Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const> const&, Eigen::internal::scalar_sum_op<double> const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x4F845B9: Eigen::internal::isApprox_selector<Eigen::Matrix<double, -1, -1, 1, -1, -1>, Eigen::Matrix<double, -1, -1, 1, -1, -1>, false>::run(Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, double const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522ACB3: ACADO::ExportNLPSolver::setLSQObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522DDA9: ACADO::ExportNLPSolver::setObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x52055A2: ACADO::OCPexport::setup() (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x5205E2F: ACADO::OCPexport::exportCode(std::string const&, std::string const&, std::string const&, int) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x406EED: main (in /home/mkatliar/ACADOtoolkit/examples/code_generation/mpc_mhe/code_generation_pendulum_dae_nmpc)
==31318== 
==31318== Conditional jump or move depends on uninitialised value(s)
==31318==    at 0x4F842FA: Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const>, 3, 0>::run(Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const> const&, Eigen::internal::scalar_sum_op<double> const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x4F845B9: Eigen::internal::isApprox_selector<Eigen::Matrix<double, -1, -1, 1, -1, -1>, Eigen::Matrix<double, -1, -1, 1, -1, -1>, false>::run(Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, double const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522ACB3: ACADO::ExportNLPSolver::setLSQObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522DDA9: ACADO::ExportNLPSolver::setObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x52055A2: ACADO::OCPexport::setup() (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x5205E2F: ACADO::OCPexport::exportCode(std::string const&, std::string const&, std::string const&, int) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x406EED: main (in /home/mkatliar/ACADOtoolkit/examples/code_generation/mpc_mhe/code_generation_pendulum_dae_nmpc)
==31318== 
==31318== Use of uninitialised value of size 8
==31318==    at 0x4F84300: Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const>, 3, 0>::run(Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, -1, -1, 1, -1, -1> const> const&, Eigen::internal::scalar_sum_op<double> const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x4F845B9: Eigen::internal::isApprox_selector<Eigen::Matrix<double, -1, -1, 1, -1, -1>, Eigen::Matrix<double, -1, -1, 1, -1, -1>, false>::run(Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, Eigen::Matrix<double, -1, -1, 1, -1, -1> const&, double const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522ACB3: ACADO::ExportNLPSolver::setLSQObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
==31318==    by 0x522DDA9: ACADO::ExportNLPSolver::setObjective(ACADO::Objective const&) (in /home/mkatliar/ACADOtoolkit/build/lib/libacado_toolkit_s.so.1.2.1beta)
[... cut ...]
==31318==      possibly lost: 64,026 bytes in 1,521 blocks
==31318==    still reachable: 488,146 bytes in 7,229 blocks
==31318==         suppressed: 0 bytes in 0 blocks
==31318== Rerun with --leak-check=full to see details of leaked memory
==31318== 
==31318== For counts of detected and suppressed errors, rerun with: -v
==31318== Use --track-origins=yes to see where uninitialised values come from
==31318== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

mvukov, 2015-03-16:
Interesting. You can try to replace Eigen with the newer one in the `external_packages` folder. I've never seen this issue before. Or, it may be we forgot to copy an object in the Objective(?) class copy-ctor... (just brainstorming)