acado/acado #141: acado.OCPexport.exportCode() outputs error message but does not throw

mkotlyar, 2015-03-09:
To reproduce: place the attached "reproduce.m" in `ACADOtoolkit/interfaces/matlab/examples/code_generation/nmpc/crane` and run it.

Output:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'g++'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
[ACADO] Error: An option with this name does not exist
  Code: (RET_OPTION_DOESNT_EXIST) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/user_interaction/options_list.hpp
  Line: 293

ACADO Toolkit::Code Generation Tool
Copyright (C) 2008-2014 by Boris Houska, Hans Joachim Ferreau,
Milan Vukov, Rien Quirynen, KU Leuven.
Developed within the Optimization in Engineering Center (OPTEC)
under supervision of Moritz Diehl. All rights reserved.

ACADO Toolkit is distributed under the terms of the GNU Lesser
General Public License 3 in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Lesser General Public License for more details.

Warning: The show does go on! 
> In reproduce_bug at 89 
```

One can see the error message from `exportCode`, but no exception is thrown and the execution continues. The error is just ignored.

rienq, 2015-03-09:
Hey Misha,
Thanks for pointing this out, I made a quick fix for this.. you can let me know what you think.
Cheers

mkotlyar, 2015-03-09:
I pulled the changes from master, rebuilt ACADO and get exactly the same issue.

rienq, 2015-03-09:
Hmm, I see.. that's bad because it means that exportCode is not raising an error at all.. the error is ignored somewhere inside ACADO then..
I can have a look at this specific case where you define a discrete time integrator for a continuous time system

rienq, 2015-03-09:
I found out what is going on for this example code.. LEVENBERG_MARQUARDT is an option which is defined in a way that it accepts a double which makes total sense since you usually should use values like "1e-8" or maybe "1e-6".. what happened here is that you defined 1e+4 which is an integer and therefore the option was not found when calling "set" in the OptionsList class.. ;-)

closed 2015-03-09 by rienq

mkotlyar, 2015-03-11:
The issue can be still reproduced with the following example:
```
clc;
clear all;
close all;

Ts = 0.05;

Nu = 8;
Nx = 2 * Nu;
Ny = 6 + Nu;

DifferentialState q(Nu) v(Nu);
Control u(Nu);

%% Differential Equation
f_discr = acado.DiscretizedDifferentialEquation(Ts);
expr_discr = next([q;v]) == [q + v * Ts + u * Ts^2 / 2; v + Ts * u];
for i = 1:length(expr_discr)
    f_discr.add(expr_discr(i));
end

%% SIMexport
acadoSet('problemname', 'sim');

sim = acado.SIMexport( Ts );
sim.setModel(f_discr);
sim.set( 'INTEGRATOR_TYPE',             'INT_DT' );
sim.set( 'NUM_INTEGRATOR_STEPS',        2        );

sim.exportCode( 'export_SIM' );
```

Output:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'g++'.
Warning: You are using gcc version '4.8.3'. The version of gcc is not supported. The version currently supported with MEX is
'4.7.x'. For a list of currently supported compilers see: http://www.mathworks.com/support/compilers/current_release. 
> In AcadoMatlab.generateCPP at 102
  In GEN_ACADO at 41
  In ExportModule>ExportModule.exportCode at 90
  In reproduce at 29 
Warning: You are using gcc version '4.8.3-12ubuntu3)'. The version of gcc is not supported. The version currently supported
with MEX is '4.7.x'. For a list of currently supported compilers see:
http://www.mathworks.com/support/compilers/current_release. 
> In AcadoMatlab.generateCPP at 102
  In GEN_ACADO at 41
  In ExportModule>ExportModule.exportCode at 90
  In reproduce at 29 
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
[ACADO] Error: No discrete-time ODEs supported for code generation
  Code: (RET_NO_DISCRETE_ODE_FOR_CODE_EXPORT) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 307
  Code: (returnvalue) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 250

[ACADO] Error: Unable to generate code
  Code: (RET_UNABLE_TO_EXPORT_CODE) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 126

>> 
```

Interestingly, on Windows I don't get any error:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'Microsoft Visual C++ 2013 Professional'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
>> 
```

rienq, 2015-03-11:
Hmm, this does not look like the same issue anymore right? And on Windows, I have indeed experienced a problem with (error) messages being printed out to the command line of MATLAB.. this is again a separate issue I would say.

mkotlyar, 2015-03-11:
There are actually two issues:
1. The code generator outputs error message and continues. This is the same issue.
2. For the case give, the code generator fails on Linux and succeeds on Windows. This is a separate issue.

mkotlyar, 2015-03-11:
After your fix, I really see it does not print error messages to MATLAB console on Windows, but the `exportCode()` throws:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'Microsoft Visual C++ 2013 Professional'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
Error using sim_RUN
ACADO export failed because of the above error(s)!

Error in run (line 63)
evalin('caller', [script ';']);

Error in acado.ExportModule/exportCode (line 93)
                run(sprintf('%s_RUN', ACADO_.helper.problemname));

Error in reproduce (line 29)
sim.exportCode( 'export_SIM' );
```
Which function do you use to output the `[ACADO] Error ...` messages?

mkotlyar, 2015-03-11:
Created #145 to address the later.