acado/acado #142: Insufficient diagnostic information from acado.OCPexport.exportCode() (MATLAB interface, master branch)

mkotlyar, 2015-03-09:
To reproduce, run the following code:
```
% A modified version of crane.m to reproduce the issue.

clc;
clear all;
close all;

Ts = 0.1;
EXPORT = 1;

DifferentialState xC vC xL vL uC uL theta omega;
Control duC duL;

n_XD = length(diffStates);
n_U = length(controls);

tau1 = 0.012790605943772;   a1   = 0.047418203070092;
tau2 = 0.024695192379264;   a2   = 0.034087337273386;
g = 9.81;
m = 1318.0;
c = 0;

%% Differential Equation
aC = is(-1.0/tau1*vC + a1/tau1*uC);
aL = is(-1.0/tau2*vL + a2/tau2*uL);

M1 = eye(6);
A1 = zeros(6,6);
B1 = zeros(6,2);
A1(1,2) = 1;
A1(2,2) = -1.0/tau1;
A1(2,5) = a1/tau1;
A1(3,4) = 1.0;
A1(4,4) = -1.0/tau2;
A1(4,6) = a2/tau2;

B1(5,1) = 1;
B1(6,2) = 1;

f = dot([theta; omega]) == [ omega; ...
    1.0/xL*(-g*sin(theta) - aC*cos(theta) - 2*vL*omega - c*omega/(m*xL)) ];

h = [diffStates; controls];
hN = [diffStates];

%% MPCexport
acadoSet('problemname', 'mpc');

N = 10;
ocp = acado.OCP( 0.0, N*Ts, N );

W_mat = eye(n_XD+n_U,n_XD+n_U);
WN_mat = eye(n_XD,n_XD);
W = acado.BMatrix(W_mat);
WN = acado.BMatrix(WN_mat);

ocp.minimizeLSQ( W, h );
ocp.minimizeLSQEndTerm( WN, hN );

Slx = [0 0 0 0 0 0 0.01 0];
Slu = [0 0];
ocp.minimizeLSQLinearTerms( Slx, Slu );

ocp.subjectTo( -10.0 <= [uC;uL] <= 10.0 );
ocp.subjectTo( -100.0 <= [duC;duL] <= 100.0 );
ocp.subjectTo( -0.3 <= vC <= 0.3 );
ocp.setLinearInput(M1,A1,B1);
ocp.setModel(f);

mpc = acado.OCPexport( ocp );
mpc.set( 'HESSIAN_APPROXIMATION',       'GAUSS_NEWTON'      );
mpc.set( 'DISCRETIZATION_TYPE',         'MULTIPLE_SHOOTING' );
mpc.set( 'QP_SOLVER',                   'QP_QPOASES'    	);
mpc.set( 'SPARSE_QP_SOLUTION',          'FULL_CONDENSING_N2');
mpc.set( 'INTEGRATOR_TYPE',             'INT_DT' );
mpc.set( 'NUM_INTEGRATOR_STEPS',        N                   );
mpc.set( 'LEVENBERG_MARQUARDT', 		 1e+4				);
mpc.set( 'MAX_NUM_QP_ITERATIONS',       200                 );

mpc.exportCode( 'export_MPC' );
```
Output:
```
[ACADO] Error: An option with this name does not exist
  Code: (RET_OPTION_DOESNT_EXIST) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/user_interaction/options_list.hpp
  Line: 293
```
From this information, there is no way to figure out which option name does not exist. I propose to add the option name to the output.

rienq, 2015-03-09:
I totally agree with you, but this might be a more complicated issue than it might seem.. because it occurs at more than one location in the ACADO code that this "invalid option" error is thrown without more information which is indeed frustrating. Milan did a huge effort once in improving this but it appears that it can still happen sometimes so the only fix would be to carefully clean up the whole ACADO code generation tool in this respect..
Currently, the best way to figure out where something went wrong is to cut down your example (e.g. remove constraints etc) until you only remain the core of your problem formulation which should get exported fine and then add your complexities one by one.. it seems annoying but it often is the quickest way to find out where something is wrong.

In this case, however, it seems obvious to me what is wrong since you defined a "discrete time integrator" by 
mpc.set( 'INTEGRATOR_TYPE',             'INT_DT' );
while you have a continuous time system

rienq, 2015-03-11:
see #141

closed 2015-03-11 by rienq