acado/acado #144: A not-existing file is included in mex build with QP_SOLVER=QP_FORCES (MATLAB interface, master branch)

mkotlyar, 2015-03-11:
To reproduce the issue, run the following code from `/ACADOtoolkit/interfaces/matlab/examples/code_generation/nmpc/crane`:
```
% This is a modified version of crane.m to reproduce the issue.

clc;
clear all;
close all;
ro
Ts = 0.1;
EXPORT = 1;

DifferentialState xC vC xL vL uC uL theta omega;
Control duC duL;

n_XD = length(diffStates);
n_U = length(controls);

tau1 = 0.012790605943772;   a1   = 0.047418203070092;
tau2 = 0.024695192379264;   a2   = 0.034087337273386;
g = 9.81;
m = 1318.0;
c = 0;

%% Differential Equation
aC = is(-1.0/tau1*vC + a1/tau1*uC);
aL = is(-1.0/tau2*vL + a2/tau2*uL);

M1 = eye(6);
A1 = zeros(6,6);
B1 = zeros(6,2);
A1(1,2) = 1;
A1(2,2) = -1.0/tau1;
A1(2,5) = a1/tau1;
A1(3,4) = 1.0;
A1(4,4) = -1.0/tau2;
A1(4,6) = a2/tau2;

B1(5,1) = 1;
B1(6,2) = 1;

f = dot([theta; omega]) == [ omega; ...
    1.0/xL*(-g*sin(theta) - aC*cos(theta) - 2*vL*omega - c*omega/(m*xL)) ];

h = [diffStates; controls];
hN = [diffStates];

%% MPCexport
acadoSet('problemname', 'mpc');

N = 10;
ocp = acado.OCP( 0.0, N*Ts, N );

W_mat = eye(n_XD+n_U,n_XD+n_U);
WN_mat = eye(n_XD,n_XD);
W = acado.BMatrix(W_mat);
WN = acado.BMatrix(WN_mat);

ocp.minimizeLSQ( W, h );
ocp.minimizeLSQEndTerm( WN, hN );

Slx = [0 0 0 0 0 0 0.01 0];
Slu = [0 0];
ocp.minimizeLSQLinearTerms( Slx, Slu );

ocp.subjectTo( -10.0 <= [uC;uL] <= 10.0 );
ocp.subjectTo( -100.0 <= [duC;duL] <= 100.0 );
ocp.subjectTo( -0.3 <= vC <= 0.3 );
ocp.setLinearInput(M1,A1,B1);
ocp.setModel(f);

mpc = acado.OCPexport( ocp );
mpc.set( 'HESSIAN_APPROXIMATION',       'GAUSS_NEWTON'      );
mpc.set( 'DISCRETIZATION_TYPE',         'MULTIPLE_SHOOTING' );
mpc.set( 'SPARSE_QP_SOLUTION',          'SPARSE_SOLVER');
mpc.set( 'INTEGRATOR_TYPE',             'INT_IRK_GL4'       );
mpc.set( 'NUM_INTEGRATOR_STEPS',        2*N                 );
mpc.set( 'QP_SOLVER',                   'QP_FORCES'    	);
mpc.set( 'HOTSTART_QP',                 'NO'             	);
mpc.set( 'LEVENBERG_MARQUARDT', 		 1e-10				);

if EXPORT
    mpc.exportCode( 'export_MPC' );
    
    cd export_MPC
    make_acado_solver('../acado_MPCstep')
    cd ..
end
```

Output:
```
...
gcc: error: ./forces/obj/forces.o: No such file or directory

MEX completed successfully.
done! --> ../acado_MPCstep.mexa64
finished mexing ../acado_MPCstep.mexa64!
```
The command issued from within `make_acado_solver()` is
```
mex -O -I. -I'.' -I'./forces/include' -lrt  -D__MATLAB__ -O ./acado_solver_mex.c ./acado_solver.c ./acado_integrator.c ./acado_auxiliary_functions.c  ./forces/obj/forces.o  -output ../acado_MPCstep.mexa64
```
but there is no such file as `./forces/obj/forces.o`.

rienq, 2015-03-11:
The reason for this is the following announcement from ETH: http://forces.ethz.ch/
Therefore, the ACADO support for FORCES is now limited to the Embotech Pro version: https://www.embotech.com/FORCES-Pro

However, since the free ETH version is still temporarily available at the moment.. you could change the line "forces/obj/forces.o" in the make script to "forces/src/forces.c". The reason for this is that FORCES does not generate source files anymore but directly provides you with the compiled code.

closed 2015-03-11 by rienq

mkotlyar, 2015-03-11:
Ok, I see.
The disadvantage of changing the script manually is that if I make a change which affects the generated script, I need to redo the change. Which means I can not make a change and just re-run everything -- I need to not to forget to make the correction to the make script.
What about making the generated code such that it checks for `forces.c` and `forces.o` and adds whichever is present/needed?

rienq, 2015-03-11:
True, but you can also make the change, rename the make script and call the renamed script to compile the generated code instead.. 
I am sorry for this inconvenience but there is essentially no guarantee that the free version of FORCES will still be available tomorrow for example