acado/acado #145: MATLAB interface (master): No diagnostic output from code-generating mex-function on Windows

mkotlyar, 2015-03-11:
To reproduce, run the following MATLAB code:
```
clc;
clear all;
close all;

Ts = 0.05;

Nu = 8;
Nx = 2 * Nu;
Ny = 6 + Nu;

DifferentialState q(Nu) v(Nu);
Control u(Nu);

%% Differential Equation
f_discr = acado.DiscretizedDifferentialEquation(Ts);
expr_discr = next([q;v]) == [q + v * Ts + u * Ts^2 / 2; v + Ts * u];
for i = 1:length(expr_discr)
    f_discr.add(expr_discr(i));
end

%% SIMexport
acadoSet('problemname', 'sim');

sim = acado.SIMexport( Ts );
sim.setModel(f_discr);
sim.set( 'INTEGRATOR_TYPE',             'INT_DT' );
sim.set( 'NUM_INTEGRATOR_STEPS',        2        );

sim.exportCode( 'export_SIM' );
```
On Linux, it outputs:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'g++'.
Warning: You are using gcc version '4.8.3'. The version of gcc is not supported. The version currently supported
with MEX is '4.7.x'. For a list of currently supported compilers see:
http://www.mathworks.com/support/compilers/current_release. 
> In AcadoMatlab.generateCPP at 102
  In GEN_ACADO at 41
  In ExportModule>ExportModule.exportCode at 90 
Warning: You are using gcc version '4.8.3-12ubuntu3)'. The version of gcc is not supported. The version currently
supported with MEX is '4.7.x'. For a list of currently supported compilers see:
http://www.mathworks.com/support/compilers/current_release. 
> In AcadoMatlab.generateCPP at 102
  In GEN_ACADO at 41
  In ExportModule>ExportModule.exportCode at 90 
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
[ACADO] Error: No discrete-time ODEs supported for code generation
  Code: (RET_NO_DISCRETE_ODE_FOR_CODE_EXPORT) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 307
  Code: (returnvalue) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 250

[ACADO] Error: Unable to generate code
  Code: (RET_UNABLE_TO_EXPORT_CODE) 
  File: /home/mkatliar/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/sim_export.cpp
  Line: 126

Error using sim_RUN
ACADO export failed because of the above error(s)!

Error in run (line 63)
evalin('caller', [script ';']);

Error in acado.ExportModule/exportCode (line 93)
                run(sprintf('%s_RUN', ACADO_.helper.problemname));
```
On Windows, it outputs:
```
Writing c++ files... 
Compiling c++ files... 
Building with 'Microsoft Visual C++ 2013 Professional'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
Error using sim_RUN
ACADO export failed because of the above error(s)!

Error in run (line 63)
evalin('caller', [script ';']);

Error in acado.ExportModule/exportCode (line 93)
                run(sprintf('%s_RUN', ACADO_.helper.problemname));
```
On Windows it says "ACADO export failed because of the above error(s)!", but the errors are not displayed.

Apparently, whatever goes to stderr and stdout appears in MATLAB console on Linux but not on Windows:
```
// print.cpp
#include <mex.h>
#include <iostream>

void mexFunction( int nlhs, mxArray *plhs[], int nrhs, const mxArray *prhs[] )
{
    std::cout << "cout: Hello world!" << std::endl;
	std::clog << "clog: Hello world!" << std::endl;
	std::cerr << "cerr: Hello world!" << std::endl;
}
```
```
mex print.cpp
print
```
On Linux, outputs:
```
cout: Hello world!
cerr: Hello world! (in red)
```
On Windows, outputs nothing.

rienq, 2015-03-11:
Hmm, that's indeed very weird! And I suppose using "printf" instead of std::cout does work because MATLAB redefines that one, no?
I am not sure what would be the best way to solve this since std::cout for example is basically used all through ACADO.. I guess all of this is caused by a cleanup Milan did a while ago, you can find more info here #66

mkotlyar, 2015-03-11:
How is error reporting/logging in ACADO organized?
A good solution would be to supply a "logging service" to the functions/classes that do logging. The solution that requires minimal changes is to use std::ostream as the logging service interface and have a global logging object, which can be get and set:
```
namespace
{
    std::ostream * _logger;
}

std::ostream& getLogger() 
{ 
    if (!_logger)
        _logger = &std::clog;
    return *_logger; 
}

void setLogger(std::ostream& logger) { _logger = &logger; }

void MyFunction()
{
    getLogger() << "MyFunction was called" << std::endl;
}

void main(int, char **)
{
    // In executables, we want messages to be logged to clog
    setLogger(&std::clog);
}

class MexPrintfLogger : public std::ostream
{
    // This class works as normal std::ostream, but writes to Matlab console.
    // In fact, if I remember correctly, it is enough to create an implementation of std::basic_streambuf<> which would call mexPrintf(), create an std::ostream and set it's rdbuf() to our customized streambuf().
}

void mexFunction( int nlhs, mxArray *plhs[], int nrhs, const mxArray *prhs[] )
{
    // In mex files, we want messages to be logged to Matlab console, using mexPrintf()
    MexPrintfLogger matlab_logger;
    setLogger(&matlab_logger);
    MyFunction(); // Will use matlab_logger.
}
```

mkotlyar, 2015-03-11:
The solution above is unsafe, because if the logger object goes out of scope `getLogger()` will return a pointer to a non-existing object. A safe solution would use reference-counting pointers. But it is a technicality.

mkotlyar, 2015-03-11:
I was thinking, there may be a nicer way to redirect output to MATLAB console, without changing the code. Do you know how does MATLAB do it?

Interestingly, `printf()` in a mex-file does output to MATLAB console of Windows, but `std::cout` does not. I was curious if the use the same API on low-level. In Windows console executable, they both call `kernel32.dll!WriteFile`, which in turn calls `kernel32.dll!WriteConsole`.

mkotlyar, 2015-03-11:
I need to check what is called on the low-level when `printf()` and `std::cout` are used from a mex dll.

mkotlyar, 2015-03-11:
This is how other people struggle with the same issue: https://www.mathworks.com/matlabcentral/newsreader/view_thread/235382

rienq, 2015-03-11:
Hey Misha,
There are indeed a few possibilities like that, all with their own (dis-)advantages.. it would be nice to have some solution which allows us to keep on using std::cout since we would not have to go through all of the source files + this would be more error-proof later on as well..
Otherwise, one could replace the occurrences of std::cout by a stringstream for example and then use "printf" instead to output it.. for errors etc this would be rather easy to fix, since I believe it only means we would have to change the 3 functions at the end of the file "acado_message_handling.cpp" but it gets more extensive with other kinds of output

mkotlyar, 2015-03-11:
`Using cout or the C-language printf function does not work as expected in C++ MEX-files. Use the mexPrintf function instead.`:
http://de.mathworks.com/help/matlab/matlab_external/creating-c-mex-files.html

mkotlyar, 2015-03-11:
This is what MathWorks says: "Using cout or the C-language printf function does not work as expected in C++ MEX-files. Use the mexPrintf function instead."
http://de.mathworks.com/help/matlab/matlab_external/creating-c-mex-files.html

rienq, 2015-03-11:
Yea, I read that as well.. I guess the best solution is to recreate some kind of "acadoPrintf" functionality where we can differentiate between MATLAB and C++ installations of ACADO, similar to what we had before the STLification modifications

mkotlyar, 2015-03-11:
What about just replacing std::cout by acado::getLogger()?

rienq, 2015-03-11:
sure, it's just a bit more tedious to actually define the new ostream.. but I guess it is not too hard, maybe something like described here:
http://stackoverflow.com/questions/243696/correctly-over-loading-a-stringbuf-to-replace-cout-in-a-matlab-mex-file

and then we could in principle even define

#ifdef __MATLAB__
   static myostream mycout;
#else 
   typedef std::cout mycout;
#end

mkotlyar, 2015-03-11:
Here is the solution, I have just tested it:
```
#include <iostream>
#include <mex.h>

class MyStreamBuf : public std::basic_streambuf<char>
{
protected:
	int_type overflow( int_type ch = traits_type::eof() ) override
	{
        if (!traits_type::eq_int_type(ch, traits_type::eof()))
            return mexPrintf("%c", traits_type::to_char_type(ch)) > 0 ? 0 : traits_type::eof();
        
        return 0;
	}
};

void mexFunction(int nlhs, mxArray *plhs[], int nrhs, const mxArray *prhs[])
{
	MyStreamBuf mybuf;
	std::cout.rdbuf(&mybuf);
	std::cout << "Hello world!" << std::endl;
}
```

rienq, 2015-03-12:
That sounds great, thanks ! :-)
I pushed it to my personal branch but I would prefer to try it out on a Windows system before I push it to the master branch, or you can also let me know if everything still compiles and works as expected if you want.
Thanks again for the help,
Rien

mkotlyar, 2015-03-12:
I keep experimenting... on my home computer (Win8 x64 + MATLAB R2013a + VS2010) std::cout and std::cerr work from mex without any additional tricks, as on Linux.
On work PC (Win7 x64 + MATLAB R2014b + VS2012) nothing is outputted to the console.
Very weird!

mkotlyar, 2015-03-12:
I will check tomorrow on my windows pc if the fix works for the given example and let you know.

rienq, 2015-03-12:
thanks! :+1:

mkotlyar, 2015-03-12:
I tested it on Windows with the example above, and it works:
```
>> mex -O  -I. -I'e:\software\ACADOtoolkit\interfaces\matlab/bin' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../acado' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../external_packages' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../external_packages/qpOASES-3.0beta/include' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../external_packages/qpOASES-3.0beta/include/qpOASES/extras' -I'e:\software\ACADOtoolkit\interfaces\matlab/../../external_packages/eigen3' -I'e:\software\ACADOtoolkit\interfaces\matlab' -DWIN32 -D__cpluplus -D__MATLAB__ -Dsnprintf=_snprintf -Dround=acadoRound -O  sim.cpp e:\software\ACADOtoolkit\interfaces\matlab\bin/acado/*.obj e:\software\ACADOtoolkit\interfaces\matlab\bin/acado/casadi/*.obj e:\software\ACADOtoolkit\interfaces\matlab\bin/acado/qpOASES/*.obj -output sim_RUN.mexw64
Building with 'Microsoft Visual C++ 2013 Professional'.
MEX completed successfully.
>> sim_RUN()

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
[ACADO] Error: No discrete-time ODEs supported for code generation
  Code: (RET_NO_DISCRETE_ODE_FOR_CODE_EXPORT) 
  File: e:\software\ACADOtoolkit\acado\code_generation\sim_export.cpp
  Line: 307
  Code: (returnvalue) 
  File: e:\software\ACADOtoolkit\acado\code_generation\sim_export.cpp
  Line: 250

[ACADO] Error: Unable to generate code
  Code: (RET_UNABLE_TO_EXPORT_CODE) 
  File: e:\software\ACADOtoolkit\acado\code_generation\sim_export.cpp
  Line: 126

Error using sim_RUN
ACADO export failed because of the above error(s)!
```

The code:
```
#include <iostream>

// Writes character stream to MATLAB console.
class MatlabConsoleStreamBuf : public std::basic_streambuf<char>
{
protected:
    int_type overflow( int_type ch = traits_type::eof() ) override
    {
        if (!traits_type::eq_int_type(ch, traits_type::eof()))
            return mexPrintf("%c", traits_type::to_char_type(ch)) > 0 ? 0 : traits_type::eof();

        return 0;
    }
};

// Automatically restores previous rdbuf() when goes out of scope.
class RedirectStream
{
public:
    RedirectStream(std::ostream& stream, std::streambuf& new_streambuf)
    :   _stream(stream)
    ,   _old_streambuf(stream.rdbuf())
    {
        _stream.rdbuf(&new_streambuf);
    }
    
    ~RedirectStream()
    {
        _stream.rdbuf(_old_streambuf);
    }
                
private:
    std::ostream& _stream;
    std::streambuf * _old_streambuf;
};
//....
void mexFunction( int nlhs, mxArray *plhs[], int nrhs, const mxArray *prhs[] ) 
 { 
    MatlabConsoleStreamBuf mybuf;
    RedirectStream redirect(std::cout, mybuf);
    //....
}
```

mvukov, 2015-03-13:
Good stuff -- @mkotlyar suggested good stuff. Redirecting std stream is the way to go, IMO. @jaeandersson did something similar for casadi-matlab interface, he might give some thoughts here.

We deliberately abandoned `acadoPrintf` crap long time ago -- and I believe we should not return to that path again.

jaeandersson, 2015-03-13:
Right, here is CasADi's implementation:
https://github.com/casadi/casadi/blob/develop/swig/matlab/meta_matlab.i