acado/acado #147: Export the Code Generator file together with the Exported Code

rafaelsaback, 2015-03-16:
I'd like to suggest that when the code generation is used, there should be an option to export also the source code used for the code generation.

Let me try to make myself clearer. Let's suppose I've created a source code called mpc_code_generation.cpp inside Acado's environment. This code will be used for exporting the controller to outside Acado's environment. It would be interesting if there was an option that makes it possible to export the mpc_code_generation.cpp together with the exported code.

I think this is interisting because inside the exported code there's no easy way of seeing which weighing matrices, horizon and other properties of the controller were used. If we export the generation code we could easily check this configurations inside it.

mvukov, 2015-03-16:
You can see a set of configuration options in `acado_common.h` (exported). It is fairly well documented.

If you would like to change the weighting matrices online, you can define them with `BMatrix` -- afterwards you will see `acadoVariables.W` and `acadoVariables.WN` in the generated header. Take a look at the examples how to use this option.

However, I still don't understand your first question nor the title of the issue...

rafaelsaback, 2015-03-16:
You're right Milan, the acado_common.h is fairly well documented, but I wasn't quite clear in my explanation. Let's consider two different things:

1) Code Generator - 1 (one) source file that is used to generate an exported controller
2) Exported Code - many source files that contains the Controller Implementation

What I really meant is that not all single information that is defined in the Code Generator is easily available in the Exported Code (even inside acado_common.h). For example, I can't see what was the dynamic model used inside Exported Code (equations and numerical values). The acado_common.h has a lot of information, but it doesn't have every single detail that is avaible at the Code Generator. So, that's why I think that it would be interesting to be able to export the Code Generator source file together with the Controller Implementation inside the Exported Code, to make it bit clearer:

Exported Code = Code Generator (1 source file) + Controller Implementation (many files)

In my case, I have manually copied the Code Generator source file inside the Exported Code folder, so I can just save it in a repository together with the Exported Code. In the future, if I want to re-generate the controller inside Acado's environment, I can just get the Code Generator from the repository and modify it to generate a new controller. It's just a matter of not losing track of which Code Generator is responsible for generating that specific Exported Code.

Kind Regards.