acado/acado #150: Accessing disturbance variable in generated code

ozymandias90, 2015-03-28:
Hi, 
I'm using an expression like this in my code:

Disturbance R;
f << 0 == dot( x ) - dx+R;

I'm trying to solve a NMPC problem - I exported the code using OCPexport. 

Now I'm trying to figure out how to access the R variable in the exported code during real time iterations, so I can update it according to another algorithm (outside ACADO). I checked the acadoVariables structure in acado_common.h but nothing was defined there.

Can you help me out?
Thanks!

rienq, 2015-04-28:
Dear Abhi,

Disturbance variables are indeed not accepted in the ACADO code generation tool. Instead, you can use OnlineData variables for your purpose so the code would look:

<code>DifferentialState x,y;
Control u;
OnlineData R1, R2;
f << dot( x ) == x/(x^2+y^2)u+R1;
f << dot( y ) == y/(x^2+y^2)u+R2;</code>

I hope this can help you further.

Best,
Rien

closed 2015-04-28 by rienq

ozymandias90, 2015-04-28:
Thanks a lot, Rien! This certainly helps - I think I can figure out how to implement the disturbances. My apologies for posting the question on the issue tracker.

Abhi

rienq, 2015-04-28:
no problem, of course! glad I could help.. ;-)