acado/acado #151: CodeGeneration: SimulinkInterface isn't working correctly

simbach, 2015-03-30:
Hello everybody,

This bug occurs while using the Simulink interface of the code generation tool of ACADO.
Compiling the code and running the Simulink model created no errors.
Nevertheless the values of the output bus-object show a strange behavior:
- The first element of the state vector x stays always at the the same value of the first x0 value.
- the first element of the y changes its value even if all values of the acadoinput.data.y are set to zero.

If I run the Matlab interface the behavior of the NMPC is fine.

I'm working with Windows 8.1 and Matlab 2014a(64-bit). I run ACADO on a VM with Ubuntu on it.

Thank you in advance
Simon

mvukov, 2015-03-30:
Is this a bug in example or your own code?

simbach, 2015-03-30:
i saw this behavior of the simulink interface with the example and my own
code.

mvukov, 2015-03-30:
Which example exactly does not work as expected?

simbach, 2015-03-30:
the getting started example of the code generation - mpc_mhe folder. 

Von Samsung Mobile gesendet

<div>-------- Ursprüngliche Nachricht --------</div><div>Von: Milan Vukov <notifications@github.com> </div><div>Datum:03.30.2015  19:44  (GMT+01:00) </div><div>An: acado/acado <acado@noreply.github.com> </div><div>Cc: simbach <bachler.simon@gmail.com> </div><div>Betreff: Re: [acado] CodeGeneration: SimulinkInterface isn't working correctly  (#151) </div><div>
</div>Which example exactly does not work as expected?

rienq, 2015-05-10:
Hey @mvukov, do you know anything about this issue? Also for me the auto generated S-functions do not work as if somehow the data is passed incorrectly..

mvukov, 2015-05-10:
I don't remember I changed anything in the Simulink interface recently (> 6 months).

@rienq What means " if somehow the data is passed incorrectly"? Please give me a more detailed error description (at the moment I cannot run the example).

rienq, 2015-05-10:
I actually believe the interface never worked because of this stupid dummy variable which is in the acadoVariables (probably to make the acadoVariables struct non-empty in case of SIMexport etc).. this was kinda hard to find though! ;-)

mvukov, 2015-05-10:
Well... I see the point of the commit  537f0e6.

Well, I am quite sure the interface worked before "that stupid dummy variable". `acadodata` (Simulink related) was intentionally desinged (by me!) to resemble `ACADOvariables` structure. If that structure is changed, that it of course does not work...

The fix is dummy as well, btw.

simbach, 2015-05-12:
If DAEs are used the data still get mixed up.

closed 2015-05-14 by rienq