acado/acado #152: MexInput compile failed

chenyutao36, 2015-03-30:
Hi all,

I am using ACADO Matlab interface to do a simple parameter estimation. When I define the data measurements as MexInput, I cannot introduce time varying constraints on Control, e.x. ocp.SubjectTo(F==MexInput).

When compiling, there are a lot of warnings and errors.  Typical errors are:

\ACADOtoolkit\interfaces\matlab/../../acado/matrix_vector/matrix_vector_tools.hpp(36) : warning C4068: unknown pragma

parameter_estimation.cpp(121) : error C2593: 'operator ==' is ambiguous

\ACADOtoolkit\interfaces\matlab/../../acado/symbolic_operator/operator.hpp(161): could be 'ACADO::ConstraintComponent ACADO::operator
        ==(ACADO::VariablesGrid,const ACADO::Operator &)'

However, when I delete this command, the compile succeed.

Do anyone know how to solve this?

Thanks!

Yutao

mvukov, 2015-03-30:
@chenyutao36 check you the most fresh code and compiler. Afterwards, please report if the problem persists. Also, next time report platform, MATLAB version and compiler.

chenyutao36, 2015-03-30:
Thank you mvukov,

I'm using the latest ACADO stable brach, in which I can define time varying control constraints in general cases except with MexInput. In other words, the constraint bounds in the right hand side of the equal sign cannot be MexInput. 

I'm using MATLAB on Windows 8.1 Professional, and the compiler is Microsoft Visual C++ 2013 Professional

Thanks,

Yutao

mvukov, 2015-03-30:
Well, this indeed seems to be a bug. Regarding the MATLAB interface, @rienq can provide some help.

aurelienhilaire, 2015-10-20:
Hi, 

I'm getting the same problem using Matlab 2012a (32bit) and Microsoft Software Development Kit (SDK) 7.1 compiler. 

How did you solve the problem ? 

Thanks, 

Aurélien

gelopezal, 2017-08-31:
Hello, 

I'm having exactly the same compilation problem with the "error C2593: 'operator ==' is ambiguous", when trying to assign time-dependent parameters with ocp.subjectTo( p1 == Matrix), where matrix has the time values in its first column and the desired corresponding parameter values in the second. Here, Matrix is not numeric, but rather a MexInputMatrix. 

Is there any news regarding this issue? I'm using ACADO with Matlab 2016b and Microsoft Visual C++ 2015 Professional. 

Thank you. 

Gerardo