acado/acado #153: N2 condensing and arrival cost for MHE

rienq, 2015-03-31:
The following combination of OCPexport options is currently not yet supported:

mhe.set( 'SPARSE_QP_SOLUTION',          'CONDENSING_N2'        );
mhe.set( 'FIX_INITIAL_STATE',                   'NO'             	              );
mhe.set( 'CG_USE_ARRIVAL_COST',         'YES'                               );

The code generation tool does not return any error, but the arrival cost code is not
exported unlike the situation when using the option 'CONDENSING' instead.

chenyutao36, 2015-03-31:
Happy to know this.

I'm doing MHE in Matlab interface. Do you know how to run closed-loop simulations? The existed tutorial is based on C code and use functions like PreparationStep() and FeedbackStep(). These functions don't exist in Matlab interface. 

I try to run simulations in Matlab by initializing the input.x and input.u as the tutorial. Also input.y and input.yN is written as those in the tutorial. But I cannot get a reasonable estimation, nor arrival cost code exists.

mvukov, 2015-03-31:
@chenyutao36 Please you the forum for support -- the issue tracker is not the right place ;)