acado/acado #157: popen not defined - cygwin, windows

horschtl, 2015-04-20:
Hi,

I've tried to compile the toolkit according to http://acado.github.io/install_windows.html but failed.

"gnuplot_window.cpp:126:59: Error:'popen' was not declared in this scope"

I'm using cygwin x64 1.7.35 with gcc 4.9.2 on Win 7 x64. 

I've already tried renaming popen to _popen, alias g++ to g++ -std=gnu++0x -U__STRICT_ANSI__ according to different posts handling the same problem and many other variations. Can anybody reproduce this error or does anybody have an idea how I can solve this?

Thanks a lot!

simbach, 2015-04-21:
I had the same error and in my opinion the fastest way to get rid of this
issue is to run ACADO on a linux distribution running on a virtual machine.

sharmin1, 2015-07-14:
i am facing the same problem and struggling alot.can u help please

bsamadi, 2015-09-22:
Same issue here. I didn't find gcc4 in the installation list of Cygwin. It seems that it's obsolete.

Bu11etmagnet, 2015-11-15:
Changing *g++ -std=c++0x* to *g++ -std=gnu++0x* in the makefile (for SciTE, not acado) worked for me. Note that alias will likely not work, because if your makefile contains *-std=c++0x* , the resulting command-line will be *g++ -std=gnu++0x ... -std=c++0x* and the latter overrides the former.

chintanp, 2015-11-24:
Any updates on this ? I am getting the same error, am moving to linux, but there should be some solutions for windows. Maybe downgrading to an older version of ACADO, or something ?

jbvaningen, 2015-12-24:
I have the exact same problem.

64 bit Win7 with Cygwin, GNU 4.9.3