acado/acado #158: Unable to use conditional statements with ACADO IntermediateState variables

darienzo, 2015-05-14:
I am trying to write a simple linear interpolation script in C++ for use with the ACADO Matlab interface.  This interpolation script with be used with an ODE function I have previously written that successfully works with the ACADO integrators.  Part of this function requires a boolean comparison between an IntermediateState variable with a double.  However, no matter what order I put the two variables in (or even if I made the double also be an IntermediateState variable as well), the boolean condition fails to compile.  It seems from the code there is an operator "<=" defined that should work with IntermediateState variables, but for some reason mine is not.

In my latest attempt, I have a failure with the statement "xdata[0] >= x", where xdata[0] is a double and x is the IntermediateState:
 error C2451: conditional expression of type 'ACADO::ConstraintComponent' is illegal 
        No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called 

I am running with the latest ACADO code on Matlab 2013a with the Visual C++ 2010 compiler.  My operating system is Windows 8.1.

rienq, 2015-05-14:
Hi,
conditional statements are non smooth and therefore indeed not supported in ACADO (just like most modeling tools for optimization by the way), so you will have to reformulate your problem in another but smooth way or sometimes you just have to replace the non smooth function by a smooth approximation..

closed 2015-05-14 by rienq

darienzo, 2015-05-14:
Okay, thanks, I wasn't clear on that since I saw there was an "<=" operator
defined for ACADO expressions. Is there any way to get the value of a state
as a double for a particular point in time for use in the ODE function?
Otherwise I'll just try that smooth approximation technique.

Thanks again for getting back to me!

Liz