acado/acado #160: Missing declaration when using code generation with external model

RoVersch, 2015-06-26:
I think there is a bug in the code generating code when using it with an external model in a seperate source file, as in one of the examples:

    sim3.setModel( "model", "rhs", "rhs_jac" );

If you have an implicit integrator, as in the example, acado_common.h has a forward declaration of rhs() in it. However, using an explicit integrator, this forward declaration is missing, resulting in compile errors. The code does compile after manually adding the missing declaration.

I looked into the source code, and found a difference in the way the function declarations are added. Compare the code in irk_export.cpp:

    returnValue ImplicitRungeKuttaExport::getFunctionDeclarations(	ExportStatementBlock& declarations
														) const
	{
		declarations.addDeclaration( integrate );

		if( NX2 != NX ) 	declarations.addDeclaration( fullRhs );
		else				declarations.addDeclaration( rhs );

	    return SUCCESSFUL_RETURN;
	}
and erk_export.cpp:
    
    returnValue ExplicitRungeKuttaExport::getFunctionDeclarations(	ExportStatementBlock& declarations
															) const
	{
		declarations.addDeclaration( integrate );

		int matlabInterface;
		userInteraction->get( GENERATE_MATLAB_INTERFACE, matlabInterface );
		if (matlabInterface) {
			declarations.addDeclaration( rhs );
		}

		return SUCCESSFUL_RETURN;
	}

Do you think this is an actual bug? Or am I missing something here?
Thx,
Robin

rienq, 2015-06-30:
But what code are you compiling exactly to get problems here?
Normally, as far as I remember at the moment, the right-hand side function is not called anywhere for ERK methods.. you only need the dynamics, extended with the variational differential equations (VDE) which in this case you provide as:
sim.setModel( "model", "not_used", "rhs_VDE" );

RoVersch, 2015-07-01:
That's right, but the problem is neither gets exported to the common header (not "rhs" and not "rhs_jac")

rienq, 2015-07-01:
In that case, there is also no forward declaration of "rhs_jac" when using an implicit integrator right? So the problem is not limited to ERK methods.. 
All this will result in compiler errors depending on how you compile things with which compiler I guess. 
For example: the extern_model.cpp example in code_generation/simulation exports no forward declaration of the "rhs_jac" function but still the example compiles and runs fine as far as I know.
Do you have problems with this example as well using the same compiler etc. ?

To be honest, the code at the moment is wrong because none of the methods should export a forward declaration of any of the model related function evaluations because these forward declarations are supposed to be only the functions that users could/should call themselves.

Your problem, for your specific compiling situation, could however be fixed by adding a simple header with a declaration of these functions which you then include in your test.c file. Don't you agree?