acado/acado #166: Missing includes in generated code using ocp export with external model

tstastny, 2015-11-03:
Code generation from the MATLAB interface misses include definitions for external C models when using acado.OCP.setModel(). The functionality exists in e.g. acado.simExport through an additional input argument in the make_acado_integrator() function (see /interfaces/matlab/examples/code_generation/simulation/pendulum.m), but not in acado.OCP where it would need to be included in the make_acado_solver() function. The acado_common.h header additionally does not automatically include the rhs_jac function declaration when using the external model.

This can be demonstrated on the latest commit on master, 5108cad588c48713619c5fe0782b76ce86cec83b by running the following example script:

```matlab
clc;
clear all;
close all;

% BEGIN_ACADO;

Ts = 0.1;

% STATES - - - - - - - -
DifferentialState x;

% CONTROLS - - - - - - -
Control u;

% OUTPUTS - - - - - - -
y = x;

% lengths
N   = 10;                   % horizon
n_X = length(diffStates);   % states
n_U = length(controls);     % controls
n_Y = length(y);            % outputs

% weights
Q = eye(n_Y+n_U,n_Y+n_U);
Q = acado.BMatrix(Q);

QN = eye(n_Y,n_Y);
QN = acado.BMatrix(QN);

% objective
h = [ y; u ];

acadoSet('problemname', 'nmpc');

% optimal control problem
ocp = acado.OCP( 0.0, N*Ts, N );

% minimization
ocp.minimizeLSQ( Q, h );
ocp.minimizeLSQEndTerm( QN, y );

% external model
ocp.setModel('model', 'rhs', 'rhs_jac');
ocp.setDimensions( 1, 1, 0, 1);

% constraints
ocp.subjectTo( -3 <= u <= 3 );

% export settings
nmpc = acado.OCPexport( ocp );
nmpc.set( 'HESSIAN_APPROXIMATION',          'GAUSS_NEWTON'          );
nmpc.set( 'DISCRETIZATION_TYPE',            'MULTIPLE_SHOOTING'     );
nmpc.set( 'SPARSE_QP_SOLUTION',             'FULL_CONDENSING'    );
nmpc.set( 'INTEGRATOR_TYPE',                'INT_IRK_GL4'           );
nmpc.set( 'NUM_INTEGRATOR_STEPS',           N                       );
nmpc.set( 'QP_SOLVER',                      'QP_QPOASES'            );
nmpc.set( 'HOTSTART_QP',                    'NO'                    );
nmpc.set( 'LEVENBERG_MARQUARDT',            1e-10                   );
nmpc.set( 'GENERATE_MAKE_FILE',             'YES'                   );
nmpc.set( 'GENERATE_TEST_FILE',             'YES'                   );
nmpc.set( 'GENERATE_SIMULINK_INTERFACE',    'YES'                   );
nmpc.set( 'CG_HARDCODE_CONSTRAINT_VALUES',  'YES'                    );

% export
copyfile('/home/thomas/ACADOtoolkit/external_packages/qpoases', ...
    '/home/thomas/ACADOtoolkit/interfaces/matlab/exmod_test/export_nmpc/qpoases')
nmpc.exportCode( 'export_nmpc' );

cd export_nmpc
make_acado_solver('../acado_nmpc_step')
cd ..
```

... with model.c:
```c
#include "acado_common.h"

void rhs( const real_t *in, real_t *out ){

/* rhs */
out[0] = in[1]; // u

}

void rhs_jac( const real_t *in, real_t *out ){
        
/* jacobian */
out[0] = 0.0; // d/dx u = 0
out[1] = 1.0; // d/du u = 1

}
```

The following errors are output to the command window:
```
Error using mex
/home/thomas/ACADOtoolkit/interfaces/matlab/exmod_test/export_nmpc/./acado_integrator.c: In function ‘int integrate(real_t*,
int)’:
/home/thomas/ACADOtoolkit/interfaces/matlab/exmod_test/export_nmpc/./acado_integrator.c:141:77: error: ‘rhs_jac’ was not
declared in this scope
 rhs_jac( acadoWorkspace.rk_xxx, &(acadoWorkspace.rk_diffsTemp2[ run1 * 2 ]) );
                                                                             ^
/home/thomas/ACADOtoolkit/interfaces/matlab/exmod_test/export_nmpc/./acado_integrator.c:189:77: error: ‘rhs_jac’ was not
declared in this scope
 rhs_jac( acadoWorkspace.rk_xxx, &(acadoWorkspace.rk_diffsTemp2[ run1 * 2 ]) );
                                                                             ^
Error in make_acado_solver (line 88)
eval( sprintf(CGRecipe, fileOut, mexext) );

Error in nmpc_setup (line 71)
make_acado_solver('../acado_nmpc_step')
```

This issue can be manually bypassed through the following:

1) Export the code, i.e. acado.OCPexport.exportCode
2) Manually insert the rhs_jac declaration in the acado_common.h header (I simply put it after the existing void rhs( ... ) declaration, but I suppose it doesn't matter where it goes).
```c
/** An external function for evaluation of symbolic expressions. */
void rhs_jac(const real_t* in, real_t* out);
```
3) Manually insert the model.c file include into both make_acado_solver.m ...
```matlab
% Auto-generated files
	CGSources = [ ...
		'CGRoot/acado_solver_mex.c ' ...
		'CGRoot/acado_solver.c ' ...
		'CGRoot/acado_integrator.c ' ...
		'CGRoot/acado_auxiliary_functions.c ' ...
		'CGRoot/acado_qpoases_interface.cpp ' ...
        'CGRoot/model.c ' ...
		];
```
... and in make_acado_solver_sfunction.m
```matlab
% Define source files
acadoDef.SourceFiles = { ...
	'acado_solver.c', ...
	'acado_integrator.c', ...
	'acado_auxiliary_functions.c', ...
	'acado_solver_sfunction.c', ...
    'model.c' ...
	};
```
4) Make the solver, i.e. run make_acado_solver()

In this sequence, the compilation will finish.

Is it possible to add the additional input argument to make_acado_solver() when exporting the ocp? Also automatic generation of an acado_common.h header with rhs_jac declaration? Manually inserting like this works, but is quite annoying.

Thanks for the help!

tstastny, 2015-12-07:
Hi again, sorry to spam (its been a month now), but has anyone had a chance to look at this? I know there are probably more pressing issues, but I think this is actually something fairly quick/easy to solve (just some missing includes and possibly adding an input arg to the to the make_acado_solver(), like I suggested in the first post above).

I'm open to making the fix myself and making a pull request, but I will need to know exactly where to make the changes (somewhere in the code generation source..?) and how the code maintainers prefer this be done (as this is not my repository!). I spent the time to make a nice explanatory working example, so any help/recommendations/opinions would be very much appreciated! Thanks!

rienq, 2015-12-22:
Thanks for posting this issue.. it indeed took a long time, but could you have a look whether the above two commits fix the problem for you? Cheers, Rien

tstastny, 2015-12-24:
Works! Thanks a lot for the support!

closed 2015-12-24 by tstastny