acado/acado #173: ACADO Simulink Interface

adexcoca, 2016-01-25:
Hi,
I just started using ACADO for Matlab. I was working with NMPC entirely developed in Matlab (m-file). But I want to start using ACADO since I will implement my NMPC very soon for real system control. I have two questions:
1) after I generated NMPC control using ACADO code generation from Matlab, I want to use the controller to control the plant developed in SIMULINK. Here, I need help on how I can interface the generated NMPC control with plant model in Simulink.
2) In the NMPC model, I have parameters which need to be updated based on some thermodynamic tables. These tables are already compiled as m-files in Matlab and every time as state changes, I have to refer some parameters as a function of states. Is it possible to compile these additional Matlab files ( collections of separate code) during code generation via ACADO? I wonder ACADO only generate code from a single m-file. 

Thank you,

JannickP, 2016-11-02:
Hi adexcoca,

I have the same question as your first one.
Did you solve it?

I have the .mex64 file from the standalone ACADO MPC controller. How can I convert it into a Simulink block? 

Thank you in advance,
Jannick

anschwarz, 2017-01-25:
Hi,
I don't know if there's a better way to do it, but I solved it the following way: 

In the Simulink model I created a MATLAB Function block which calls a MATLAB function out of the Workspace which calls the ACADO controller. Sounds complicated, but the workaround is needed because Simulink usually only works with some matlab commands which are supported by the code generation feature. I couldnt find a solution to call the generated ACADO m-file directly.

Example for the MATLAB function in a Simulink block (In the block parameters you can specify the sample time of the controller):

`function u = acado_vehicle_controller( x0, a_setpoint )
coder.extrinsic('acado_vehicle_mpc_call');
u = coder.nullcopy(0);
u = acado_vehicle_mpc_call( x0, a_setpoint );
end`
The `coder.extrinsic` command is necessary to tell Simulink not to generate C-code from the passed function. The `coder.nullcopy` command is sometimes necessary for Simulink to determine the right signal dimensions (I ran into lots of errors. Sometimes it worked without the nullcopy, sometimes not. I don't really get the point...). Sometimes I also needed to fix the signal dimensions in the model explorer.

Example for the second call function:

`function u = acado_vehicle_mpc_call( x0, a_setpoint )
refMatrix = [0 a_setpoint 0]; 
out = vehicle_RUN( 0, x0, refMatrix);
u = out.U(1);
end`

Hope this helps someone!
Andreas

adexcoca, 2017-01-25:
Thank you, I able to integrate by generating  s-function. In ACADO code generation m-file u just put a line code that generate an S-function. 

Are you able to generate a code from Simulink for hardware implementation? That is where I have difficulty this time.

anschwarz, 2017-01-25:
No I don't face this problem because I am only doing simulations. But I think the best bet would be to use the automatic code generation tool of the (original - not MATLAB) ACADO Toolkit.

rienq, 2017-01-25:
I would indeed use the code generation tool in order to do NMPC in Simulink, you can even generate the S-function automatically. Even though the documentation sadly is indeed quite outdated, you can find some examples for the use of the code generation tool both in the main examples folder and in interfaces/matlab/examples if you are using the MATLAB interface (both the C++ and MATLAB interface offer the same features regarding code generation)

anschwarz, 2017-01-25:
Yeah, the examples are very nice to get started. The problem is that you don't look for examples of a feature you don't know it even exists ;)  (I just followed the manual)

Nevertheless there is one big advantage when using my workaround: You don't have to bother with all the settings, especially with integrator step size and integrator type.

adexcoca, 2017-01-25:
Hi Rieng,

I have been used ACADO code generation tool from matlab for about a year, I
able to generate s-function for simulink from m-file. All working fine for
simulation.
I even submitted a journal paper on NMPC application for heavy-duty diesel
engine waste recovery control problem using the simulation result via ACADO
code generation. Now my next task is validating the controller on actual
hardware. For this I need to generate an executable code from simulink
model with ACADO s-function for NMPC for Microautobox (dspace). I tried but
failed to generate the necessary code for some problem related to
ACADOoutput data type compatibility issue. There is no documentation and
example case for such purpose.

Do you have any comment on this issue?

rienq, 2017-01-30:
Hi there,

I am glad to hear that the ACADO code generation has been a useful tool for your research, at least in order to gain simulation results then so far. Regarding the use of the generated S-function on dSPACE: it has been a known issue that the bus signals which are used in the ACADO generated S-function are not compatible with (all versions of) dSPACE.. to my knowledge, but maybe there is a better solution, the easiest is to just use your own S-function in C similar to the file here for the robot example:
https://github.com/acado/acado/blob/master/interfaces/matlab/examples/code_generation/nmpc/robot/export_mpc/sfunction_robot.cpp
Note that you can write the same S-function also in C code, when additionally using the C version of qpOASES for example or any other QP solver in C code, because of the problems of putting C++ code on dSPACE. I hope this can more or less answer your question.

JavierPradoRomo2007, 2017-09-26:
[ControlAcado.zip](https://github.com/acado/acado/files/1334548/ControlAcado.zip)
Hi there,
I have been testing ACADO tolkit and It is a great tool, however I am working not only in the MATLAB console. I am using it in Simulink. I based my project on the example proper of ACADO as "template" (interfaces\matlab\examples\code_generation\nmpc\robot\init_sim.m)  Everything seems to work, but the control output from the controller is zero. It was tested in the console environment and my project works fine. I am using an Sfunction supported by an .cpp. I checked feedback states and control outputs, but Control outputs show as zero. Everything compiles, then I do not where to fix my problem. Can someone help me!.  I will attach my example and I print where the problem is possible. Thanks beforehand for helping me.

adexcoca, 2017-09-26:
Hey, directly send your question to my email “ ayebi@clemson.edu” so that I can help.

Adamu

ernstpre, 2022-02-15:
I would be quite interested to hear what the problem was to the issue that @JavierPradoRomo2007 had? I am having what sounds like very similar issues in trying to work with the ACADO generated Simulink interface.