acado/acado #178: Compilation failed in ACADO with Force

chenyutao36, 2016-02-11:
According to the manual, the QP solver Forces is supported in ACADO and the only modification of the code generation is to set

mpc.set( 'SPARSE_QP_SOLUTION',          'SPARSE_SOLVER');
mpc.set( 'QP_SOLVER',                   'QP_FORCES'    	);

The forces code is successfully generated from the server as well as the mex interface.
However, when compiling the code I get the following error

Error using mex
acado_solver.c
...\forces\acado_solver.c(876) : error C2198: 'forces_solve' : too few arguments for call

Error in make_acado_solver (line 81)
eval( sprintf(CGRecipe, fileOut, mexext) );

BTW, I'm using the latest version of Forces Pro.

Can someone solve this problem?

rienq, 2016-07-26:
This should be fixed in the latest master branch, is it not?

closed 2016-07-26 by rienq

chenyutao36, 2016-07-26:
Yes, this single problem is fixed