acado/acado #179: Matlab interface not compiling

clusher, 2016-02-17:
Hello,

I have installed Matlab R2015B on Windows 10. I have used mex to setup Microsoft Visual C++ 2015 Professional as compiler for C++. Then I used GIT to clone the stable branch of acado. And finally went to interfaces/matlab and ran make clean all. Unfortunately it failed with:
`Making ACADO... 
*Building with 'Microsoft Visual C++ 2015 Professional'.
Error using mex
acado_io_utils.cpp
C:\Program Files (x86)\Windows Kits\10\\include\10.0.10150.0\ucrt\stdio.h(1925): warning C4005: 'snprintf':
macro redefinition
C:\Program Files (x86)\Windows Kits\10\\include\10.0.10150.0\ucrt\stdio.h(1925): note: command-line
arguments:  see previous definition of 'snprintf'
C:\Program Files (x86)\Windows Kits\10\\include\10.0.10150.0\ucrt\stdio.h(1927): fatal error C1189: #error:
Macro definition of snprintf conflicts with Standard Library function declaration

Error in makehelper>execute_command (line 332)
    eval (s) ;
Error in makehelper (line 237)
                    execute_command (cmd, DEBUG, SRC{i}, ~PARALLEL) ;
Error in make (line 79)
    makehelper(0, {}, varargin);`

I searched for similar mistakes and found the Windows Fix at issue #172 . I applied it, but the result was the same. Any ideas?
Thank you!

rienq, 2016-02-17:
Thanks for creating this issue! The fix in issue #172 indeed does not yet fix the MATLAB interface. If you want to try out a quick fix, you can simply go to the file 'interfaces/matlab/makehelper.m'
and change this part of the code in your local repository:

%COMPILER FLAGS
    if (ispc)
        % Microsoft Visual C++ (express) compiler
        CPPFLAGS  = [ IFLAGS, ' -DWIN32 -D__cpluplus -D__MATLAB__ -Dsnprintf=_snprintf -Dround=acadoRound -O ' ];    

into the following:

%COMPILER FLAGS
    if (ispc)
        % Microsoft Visual C++ (express) compiler
        CPPFLAGS  = [ IFLAGS, ' -DWIN32 -D__cpluplus -D__MATLAB__ -Dround=acadoRound -O ' ];  

so you simply remove the redefinition of snprintf and this should do the trick.

However, I would of course like to update the makehelper.m file to add the right flags depending on the compiler version. For this purpose, could you maybe check the outcome of writing "comp = mex.getCompilerConfigurations('C++','Selected')" in MATLAB on your Windows machine and post the result here? There should be a field 'Version' which in your case I guess is 14..?

clusher, 2016-02-17:
Many thanks, I would have no idea where the snprintf definition would be. It worked ok now.

Regarding what you asked, the version is indeed 14.0. The output was:
`             Name: 'Microsoft Visual C++ 2015 Professional'
     Manufacturer: 'Microsoft'
         Language: 'C++'
          Version: '14.0'
         Location: 'C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\..\'
        ShortName: 'MSVCPP140'
         Priority: 'B'
          Details: [1x1 mex.CompilerConfigurationDetails]
       LinkerName: 'link'
    LinkerVersion: ''
           MexOpt: 'C:\Users\...\AppData\Roaming\MathWorks\MATLAB\R2015b\mex_C++_win64.xml'`

rienq, 2016-02-18:
Great, thanks! This should be fixed now in the master branch..

closed 2016-02-18 by rienq