acado/acado #188: NMPC-MHE

adexcoca987, 2016-05-17:
Hi all,

I am looking for a working code of nonlinear model predictive control with moving horizon estimator (NMPC-MHE) for Matlab implementation. I found an example case in ACADO for Matlab example folders, but couldn't run. 

Actually, I am looking specifically an example code that shows how the auto-code generation can be written in Matlab to get real time implementable NMPC solver and MHE solver, possible for Simulink interface. 

Regards,

rienq, 2016-05-18:
The issue tracker is not supposed to be used for questions regarding how to use the code but instead to report bugs or request features. A discussion forum can be found here:
https://sourceforge.net/p/acado/discussion/general/

In case one of the code generation examples did not run, based on the latest version of ACADO, you can make a separate issue for this including all details regarding your system and the obtained error messages.

Finally, some extra ACADO course material can be found here:
http://syscop.de/event/internal-acado-course-%E2%80%93-automatic-control-and-dynamic-optimization
and
http://syscop.de/event/acado-course-%E2%80%93-automatic-control-and-dynamic-optimization

closed 2016-05-18 by rienq

adexcoca987, 2016-05-26:
Hi,

Thank you for the replay.
These days I am working on moving horizon estimation(mhe) plus nonlinear
model predictive control(nmpc). I tried to run example file in the
following link but couldn't work.

acado/examples/code_generation/mpc_mhe/crane_closed_loop

Would you help me providing working version of this example file.

In matlab, if I want to generate s_function for mhe and nmpc, is it
possible to change the name of generated s_function. I tried the following
function but it couldn't work.

mhe.set( 'CG_MODULE_NAME', 'mhe');

I appreciate the help,

Thank you,

rienq, 2016-05-30:
Hey there,

this issue has been closed. If you found an actual bug, you should each time make a separate issue to specify exactly what error messages you get, on which system and running which code.. because the same example might run fine on a different platform.

Thanks,
Rien