acado/acado #189: Compilation of Forces in Windows

chenyutao36, 2016-05-19:
The compilation of Forces in Windows using VS is not working in the latest version of ACADO. (It does work in Linux using gcc). 

A feedback from Forces team claims that ACADO shoud add (or modify) the following commands in the file "make_acado_solver.m"

**ldFlags = '-lmmt -lirc -lsvml_dispmt -ldecimal';**
CGRecipe = [ ...
        'mex -O' ...
        ' -I. -I''CGRoot'' -I''CGRoot/forces/include''' ...
        ' ldFlags' **' -L''CGRoot/forces/lib'''** ...
        '  -D__MATLAB__ -DUSEMEXPRINTS -O CGSources qpFORCESSources -output %s.%s' ...
        ];

I test it successfully in Windows 10 with VS 2013 pro.

Also, I found that affine constraints cannot be generated in the file "acado_forces_generator.m". I personally ask for a fix for this.

Best,

Yutao

adexcoca987, 2016-07-26:
Thank you for the help, but I don't know where to add these lines of code, at which folder?

chenyutao36, 2016-07-26:
In the generated folder (i.e. forces_solver) there is a file called "make_acado_solver.m" in which ACADO uses this file to generate the Forces request to for its server to generate Forces solver.

adexcoca987, 2016-07-29:
Hi Yutao,

Thank you for the comment. I did followed your suggestion as follows got the error message at the end.

Part of ACADO code generation lines:

mpc = acado.OCPexport( ocp );
mpc.set( 'HESSIAN_APPROXIMATION',       'GAUSS_NEWTON'      );
mpc.set( 'DISCRETIZATION_TYPE',         'MULTIPLE_SHOOTING' );
mpc.set( 'DYNAMIC_SENSITIVITY',         'FORWARD'           );
mpc.set( 'SPARSE_QP_SOLUTION',          'SPARSE_SOLVER'     );
mpc.set( 'INTEGRATOR_TYPE',             'INT_RK4'       );  
mpc.set( 'NUM_INTEGRATOR_STEPS',        2*N            );  
mpc.set( 'QP_SOLVER',                   'QP_FORCES'    	);

if EXPORT
    mpc.exportCode( 'export_MPC' );
    cd export_MPC
    make_acado_solver('../acado_MPCstep')  
    cd ..
end

ACADO Code generation progress:

This is FORCES Pro, a code generator for fast numerical optimization.
Copyright (C) embotech GmbH [info@embotech.com], 2013-16. All rights reserved.
Using server https://www.embotech.com/codegen
Retrieving web service specs...    [OK]
Preparing data to be sent...       [OK]
Generating and compiling code...   [OK]
Code successfully generated in 16.64 sec.
Downloading generated code...      [OK]
Package downloaded to 'forces.zip'
Extracting package...              [OK]
Building MEX interface...Building with 'Microsoft Windows SDK 7.1 (C)'.
MEX completed successfully.
          [OK]
Building Simulink S-function...Building with 'Microsoft Windows SDK 7.1 (C)'.
MEX completed successfully.
Building with 'Microsoft Windows SDK 7.1 (C)'.
MEX completed successfully.
    [OK]
Cleaning up...          Restoring recycling status to off (the files deleted should still be in the bin)
[OK]
Code generation and MEXing successfully completed. Happy solving!

Your downloaded files contain the following directories:
	 - include: C header files
	 - interface: Compiled Matlab and Simulink interfaces for your solver on the host machine (multistage)
	 - obj: solver object files for simulation on your host machine
	 - src: solver source code files (only with FORCES PRO Premium)

The approximate properties of the solver code are:
	- Data memory consumption : 82552 Bytes (statically allocated)
	- Code memory consumption : ~ 35000 Bytes
	- Computation count : 137618 FLOPS per interior-point iteration
compiling... Building with 'Microsoft Windows SDK 7.1 (C)'. 

Error message at the end:
Error using mex
LINK : fatal error LNK1104: cannot open file 'libmmt.lib'

after the above error message, I opened the generated force solver folder/ "make_acado_solver.m" and changed the line of code that you suggested below and saved:

ldFlags = '-lmmt -lirc -lsvml_dispmt -ldecimal -lIphlpapi';
CGRecipe = [ ...
'mex -O' ...
' -I. -I''CGRoot'' -I''CGRoot/forces/include''' ...
' ldFlags' ' -L''CGRoot/forces/lib''' ...
' -D__MATLAB__ -DUSEMEXPRINTS -O CGSources qpFORCESSources -output %s.%s' ...
];

Questions:
1) Once I modified the  "make_acado_solver.m", what will be the next step? 
2) what you mean by " two independent steps for compiling Forces, i.e. first compile ACADO and then generate the file "acado_forces_generator.m"; then send it to Forces server and compile the code generated by Forces."

I found the "acado_forces_generator.m" in the generated FORCES solver, but I don't know how to send it to forces server and compile the code generated by Forces?

Regards,

nielsberger, 2017-08-28:
In case anyone is still struggling with this; pause your acado execution after the 'exportCode' command, then edit the 'make_acado_solver' file using the code supplied in the first post (in case it was not clear; the **bold**-printed parts need to be added, the rest is already there. Then continue execution.