acado/acado #192: Generated Integrator gives NaN output

chenyutao36, 2016-07-08:
Hi there,

I'd like to report a possible bug in the generated integrator of ACADO.

For a non-stiff model with 17 states, 3 controls and 129 parameters, the implicit integrators give NaN output occasionally.  These integrators include at least GL4 and RIIA5. Moreover, for this model, the explicit integrator like RK4 takes >3 hrs of compilation and gives bad result. 

This phenomenon is also observed when using the example "overhead crane". When using implicit integrators to do Monte-Carolo simulations, occasionally NaN output is observed.

As the integrator is so important in ACADO, I'd like to report this to the developers. Perhaps some simplifications in the integrator are responsible for this bug.

Yours,

Yutao

rienq, 2016-07-14:
Hey Yutao,

Thanks for reporting this (possible) bug! However, this problem could be inherently related to
the kind of methods that are implemented in these implicit integration schemes though. As long as
you provide inputs which are relatively close to each other, things work typically fine as is often the
case in direct optimal control. If you however have a rather nonlinear model and you call the integrator
with very different (randomly generated) inputs, it might be that the iterative procedure does not converge and you start getting NaNs as output soon. By the way, I suppose you talk about online data variables when you mentioned "parameters" for ACADO code generation?

This said, there is little that can be done without affecting the computational performance for embedded optimization. I agree though that this is important and maybe an alternative implementation can be made at some point.

Best,
Rien

chenyutao36, 2016-07-14:
Hi Rien,

Thank you for the response. 

When I say "parameters", I actually mean "online data". However, I have tried to reduce the number of parameters to only 5, but still the NaN output is observed. Indeed, I've confirmed that for the same set of inputs, CVodes and Casadi can give correct outputs.

I think your guess is quite correct:

1) I get NaN output when the nonlinear system is behaving significantly differently from the previous steps

2) for the over head crane, since I use random inputs to do Monte-Carlo simulation, this may result NaN output if two random inputs are too different.

However, I still can't understand why the integrator gives a NaN output in this case. I can understand that when two inputs are too far way, the RTI scheme or some of the optimization steps (like qpoases sequences) may fail.  But for integrator this should not be influenced.

Best,

Yutao

rienq, 2016-07-14:
Hi Yutao,

The reason is simple: essentially, also some kind of "real-time" implementation is made for the implicit integration scheme making some computational shortcuts here and there which could be causing the problem in your case. One way to "fix" or at least avoid this though is by taking more integration steps of the same or even a different collocation scheme, as this should positively affect the convergence properties of the Newton-type scheme in the implementation. 

CVodes for example does these kind of adaptations (and much more) automatically in order to avoid numerical problems within the iterative scheme.

Best,
Rien