acado/acado #193: USE_SINGLE_PRECISION

jegan1981, 2016-07-21:
Hello,

First of all , I would like to thank you for providing very useful ACADO tool kit. I am a beginner for using this acado tool. I have to generate C code for MPC algorithm and simulate in real time using a microcontroller. Therefore, I should generate C code with single precision. I am using matlab Interface.
Please refer the links for the program.

[ini_m.txt](https://github.com/acado/acado/files/375483/ini_m.txt)
[acado_export_m.txt](https://github.com/acado/acado/files/375484/acado_export_m.txt)

Initially, I developed a plant in simulink (Designed in double precision as usual) and tried to simulate the generated acado simulink object. Matlab crashed while simulating the object and showed the erro message. Please refer the link for the error message. 

[MATLAB_error_message.txt](https://github.com/acado/acado/files/375362/MATLAB_error_message.txt)

Then I developed the plant with single precision in simulink and tried to simulate the acado simulink object. ACADO Showed the following error message.

Error reported by S-function 'acado_solver_sfun' in 'mpc_i2_kont_modell_single/acado_solver_sfun':
The Bus/StructType 'ACADOinput' must have 4 elements.
Component: Simulink | Category: Block error

Also, I found that in the generated S function( by Debugging the code using visual Studio), the simulink structure for Input signal u1 and the Output signal y1 is in double precision even if i set the Option USE_SINGLE_PRECISION. But, the acado_step( __u1BUS, __y1BUS) function Parameters u1BUS and y1BUS is in single precision. Therefore the Memory copy Statements cause datatype mismatch. Please see the part of generated S function code below.

static void mdlOutputs(SimStruct *S, int_T tid)
{
  if (isDWorkNeeded(S)) {
    /*
     * Access bus/struct information
     */
    int32_T *__dtSizeInfo = (int32_T *) ssGetDWork(S, 0);
    int32_T *__dtBusInfo = (int32_T *) ssGetDWork(S, 1);

    /*
     * Get access to Parameter/Input/Output/DWork/size information
     */
    char *u1 = (char *) ssGetInputPortSignal(S, 0);
    char *y1 = (char *) ssGetOutputPortSignal(S, 0);

    /*
     * Locally declared variable(s)
     */
    ACADOinput *__u1BUS = (ACADOinput *) ssGetPWorkValue(S, 0);
    ACADOoutput *__y1BUS = (ACADOoutput *) ssGetPWorkValue(S, 1);

    /*
     * Assign the Simulink Structure u1 to the Legacy Structure __u1BUS
     */
    (void) memcpy(&__u1BUS[0].control
                  , u1 + __dtBusInfo[0], __dtBusInfo[1]);
    (void) memcpy(&__u1BUS[0].shifting
                  , u1 + __dtBusInfo[2], __dtBusInfo[3]);
    (void) memcpy(&__u1BUS[0].initialization
                  , u1 + __dtBusInfo[4], __dtBusInfo[5]);
    (void) memcpy(__u1BUS[0].data.x
                  , u1 + __dtBusInfo[6], __dtBusInfo[7]);
    (void) memcpy(__u1BUS[0].data.u
                  , u1 + __dtBusInfo[8], __dtBusInfo[9]);
    (void) memcpy(__u1BUS[0].data.y
                  , u1 + __dtBusInfo[10], __dtBusInfo[11]);
    (void) memcpy(__u1BUS[0].data.yN
                  , u1 + __dtBusInfo[12], __dtBusInfo[13]);
    (void) memcpy(__u1BUS[0].data.W
                  , u1 + __dtBusInfo[14], __dtBusInfo[15]);
    (void) memcpy(__u1BUS[0].data.WN
                  , u1 + __dtBusInfo[16], __dtBusInfo[17]);
    (void) memcpy(__u1BUS[0].data.x0
                  , u1 + __dtBusInfo[18], __dtBusInfo[19]);

    /*
     * Call the legacy code function
     */
    acado_step( __u1BUS, __y1BUS);

    /*
     * Assign the Legacy Structure __y1BUS to the Simulink Structure y1
     */
    (void) memcpy(y1 + __dtBusInfo[20], &__y1BUS[0].status
                  , __dtBusInfo[21]);
    (void) memcpy(y1 + __dtBusInfo[22], &__y1BUS[0].kktValue
                  , __dtBusInfo[23]);
    (void) memcpy(y1 + __dtBusInfo[24], &__y1BUS[0].objValue
                  , __dtBusInfo[25]);
    (void) memcpy(y1 + __dtBusInfo[26], &__y1BUS[0].execTime
                  , __dtBusInfo[27]);
    (void) memcpy(y1 + __dtBusInfo[28], __y1BUS[0].data.x
                  , __dtBusInfo[29]);
    (void) memcpy(y1 + __dtBusInfo[30], __y1BUS[0].data.u
                  , __dtBusInfo[31]);
    (void) memcpy(y1 + __dtBusInfo[32], __y1BUS[0].data.y
                  , __dtBusInfo[33]);
    (void) memcpy(y1 + __dtBusInfo[34], __y1BUS[0].data.yN
                  , __dtBusInfo[35]);
    (void) memcpy(y1 + __dtBusInfo[36], __y1BUS[0].data.W
                  , __dtBusInfo[37]);
    (void) memcpy(y1 + __dtBusInfo[38], __y1BUS[0].data.WN
                  , __dtBusInfo[39]);
    (void) memcpy(y1 + __dtBusInfo[40], __y1BUS[0].data.x0
                  , __dtBusInfo[41]);
  }
} 

Could anybody help me to solve this Problem. Thank you .

With best regards,
Jegan

rienq, 2016-07-21:
Hey Jegan,
Thanks a lot for making this issue here!
It is a bit confusing though that you seem to mixing two very different problems here so I am going to concentrate on the single precision one (the last one, for which you provide a detailed error message) so I would suggest you make a separate issue for the first error if you would like someone to have a look at that..
I already found a first bug related to creating Simulink interfaces in single precision, I just pushed it to the master branch so you can try it out by pulling the latest version.
Cheers,
Rien

jegan1981, 2016-07-22:
Dear Rein,

Thank you for your quick response. I updated  your latest modification mentioned in the link

https://github.com/acado/acado/commit/59883482a4a5577fe8c21e9c87ecc2f22ed8d8ba

The update works for me when i use "single"  instead of "float" in that particular statement. I mentioned this in that link as well.

Thank you.

with best regards,
Jegan