acado/acado #194: ACADO Code generation issue with FORCES in Matlab

adexcoca987, 2016-07-23:
Hi all,

I am working with NMPC for engine thermal dynamics. Currently, the NMPC via ACADO is controlling set point temperature well for selected engine operating but its performance is not good for transient engine operating condition. I compared the result with linear MPC under Matlab toolbox linearizing the system dynamics at every update stage. I thought changing the solver from "QPOASES" to "FORCES" or other ACADO solver may improve the NMPC performance, but couldn't able to generate the NMPC solver except for "QPOASES" solver.  Is there any thought or suggestion for to successfully generate NMPC solver for solver like "FORCES and qpDUNES). I am using matlab interface to generate the solver. 

Below is the code I implemented and the error message:
acadoSet('problemname', 'mpc');
ocp = acado.OCP( 0.0, N*Ts_mpc, N );

W_mat=eye(3);
WN_mat =eye(2);
W = acado.BMatrix(W_mat);
WN = acado.BMatrix(WN_mat);

ocp.minimizeLSQ( W, h );
ocp.minimizeLSQEndTerm( WN, hN );
ocp.subjectTo( 0.1 <= [L1_TP;L2_TP] <= 16.0 );
ocp.subjectTo( 373.0 <= [Tw1_TP;Tw2_TP;Tw3_TP] <= 673.0 );
ocp.subjectTo( 1400e3 <= [ho_TP] <= 1800e3 );
ocp.subjectTo(0.002 <= [u_TP] <= 0.0937);
ocp.subjectTo(-ub_TP <= [uL_TP] <= ub_TP);
ocp.setModel(f);

mpc = acado.OCPexport( ocp );
mpc.set( 'HESSIAN_APPROXIMATION',       'GAUSS_NEWTON'  );
mpc.set( 'DISCRETIZATION_TYPE',         'MULTIPLE_SHOOTING' );
mpc.set( 'DYNAMIC_SENSITIVITY',         'FORWARD'           );
mpc.set( 'SPARSE_QP_SOLUTION',          'SPARSE_SOLVER'   ); 
mpc.set( 'QP_SOLVER',                   'QP_FORCES'        );
mpc.set( 'INTEGRATOR_TYPE',             'INT_IRK_RIIA5'       );
mpc.set( 'NUM_INTEGRATOR_STEPS',        2*N            );

if EXPORT
    mpc.exportCode( 'export_MPC' );
   cd export_MPC
    make_acado_solver('../acado_MPCstep')
    cd ..
end

Error message:
Undefined function or variable 'MultistageProblem'.
Error in acado_forces_generator (line 117)
stages = MultistageProblem( N );
Error in make_acado_solver (line 32)
	acado_forces_generator;
Error in ORC_ACADO_NMPC_Simulink_Code_Generation_TP_U (line 238)
    make_acado_solver('../acado_MPCstep')

Regards,

rienq, 2016-07-24:
Hey,

Even though ACADO code generation has an interface to convex solvers such as FORCES, this does not mean that you don't need to obtain a license for this software first.. In case of FORCES, a license could be obtained here:
[](https://www.embotech.com/FORCES-Pro)

Note that qpDUNES instead is free and open-source and can be obtained from this repository:
[](https://github.com/jfrasch/qpDUNES)

I hope this can help you further.

Best,
Rien

closed 2016-07-24 by rienq

adexcoca987, 2016-07-24:
Thank you for the response. I did download the folder for 'qpDUNES' and trial version for "FORCES" and put under acadotoolkit folder in the "external_packages" and tried to locate the folder in the acado code generation using the following code line:

if EXPORT
    mpc.exportCode( 'export_MPC' );
    global ACADO_;
    copyfile([ACADO_.pwd '/../../external_packages/qpdunes'], 'export_MPC/qpdunes')
    cd export_MPC
    make_acado_solver('../acado_MPCstep')
    cd ..
end

I got the following error:

Error using copyfile
No matching files were found.

Error in ORC_ACADO_NMPC_Simulink_Code_Generation_TP_U (line 246)
    copyfile([ACADO_.pwd '/../../external_packages/qpdunes'], 'export_MPC/qpdunes')

The same error shows for FORCES too, but it works for:
copyfile([ACADO_.pwd '/../../external_packages/qpoases3'], 'export_MPC/qpoases3')

Do you have any suggestion?

Regards,

adexcoca987, 2016-07-24:
I try to solve the issue of code generation with "FORCES PRO", but couldn't pass the following error. I am using ACADO for Matlab interface on windows. Is there any suggestion to overcome the following error? 

Error using mex
LINK : fatal error LNK1104: cannot open file 'libmmt.lib'
Error in makeacadosolver (line 77)
eval( sprintf(CGRecipe, fileOut, mexext) );
Error in ACADOFORCES (line 83)
makeacadosolver('../MPCforces')

Regards,

rienq, 2016-07-25:
There is no need to save these solvers in the 'external_packages' folder and then using `copyfile' whenever you do code generation (even though it should work as long as you make sure that the relative path which you provide, is correct).. Instead, you could simply copy the solvers manually in the ACADO export folder and things should work as well.

Regarding FORCES, I am not very familiar with this solver, their latest developments and especially not on a Windows machine.. but you are at least able to use it without ACADO so you are sure that you set the paths right etc?

rienq, 2016-07-26:
Maybe you should have a look here: https://github.com/acado/acado/issues/189