acado/acado #197: ACADO Code generation issue with MHE in Matlab

adexcoca987, 2016-08-26:
Hi,
I am enjoying the closed-loop performance of NMPC-MHE but I have some difficulty generating ACADO code for more practical problem with process noise.

I have no clue how to inject the process noise in the model for the code generation. I also found out that the following function are not working in the current version of ACADO.

mhe.set( 'CG_COMPUTE_COVARIANCE_MATRIX',     'YES'          );
mhe.set( 'CG_USE_ARRIVAL_COST',              'YES'          );

Is there any chance that I can get any tutorial of ACADO MHE with process noise and arrival cost definition. 

Thank you,

rienq, 2016-08-29:
Hi there,

this issue becomes hard to solve if you specify the problem as "the following options are not working".. I have to close this for now until there is a more clear description of what goes wrong and in which situation, which output errors etc..

Last time I checked, theses exact same options worked for me.

By the way: if you want the process noise to be an optimization variable, just declare it as a control input.. if it should be fixed in the OCP formulation, make it into an online data variable.

Best,
Rien

closed 2016-08-29 by rienq

JXZhou88, 2017-02-22:
Hi Rien and all the other members of ACADO,

is there any documentation on how to setup the arrival cost for the MHE cop, maybe an example code where the arrival cost is implemented? 

best regards. 

Jianxin Z.

rienq, 2017-02-22:
see #82