acado/acado #198: Weighting matrix does not generate (W, WN)

olzhas, 2016-08-31:
I have a code something like this

	OCP ocp(t_start, t_horizon, 10);  
	ocp.minimizeLSQ(W, hh);
	ocp.minimizeLSQEndTerm(W_end, hh_end); 
	//...
	OCPExport mpc(ocp);
	//...
	mpc.set( GENERATE_MATLAB_INTERFACE, YES );
	mpc.set( GENERATE_SIMULINK_INTERFACE, YES );
	mpc.set( CG_USE_VARIABLE_WEIGHTING_MATRIX, YES 				);

I looked at the acado_common.h, it does not contain W or WN. also 
#define ACADO_WEIGHTING_MATRICES_TYPE 0
then I found that the type of weighting matrix is set in 
https://github.com/acado/acado/blob/a48c39bf69fa52997f6a14b821ba016f57dd0630/acado/code_generation/export_nlp_solver.cpp#L1865
so, if I understood it right if the objectives are set I cannot set a weighting matrix?

rienq, 2016-08-31:
Yea, the code line you are referring to is exactly the answer to your "issue".. if you provide fixed values for the weighting matrices then they will be fixed also in the generated code.
The way to be able to adjust the weightings in the generated code, is by defining the weighting matrices as BMatrix objects (which stands for "boolean matrix" since they define sparsity patterns of the weighting matrix instead of specific values).. you can also find this info in the manual and/or examples

The option which you set "CG_USE_VARIABLE_WEIGHTING_MATRIX" is when you additionally want the weighting matrix to be time varying, i.e. a different matrix at each node so this does not do anything when you fix the values but only has an effect for variable BMatrix weighting matrices

closed 2016-08-31 by rienq

olzhas, 2016-08-31:
Before I used DMatrix for W instead of BMatrix.
I thought that there was no difference.