acado/acado #201: Using Auto-Generated C-Code in a C++ Program for Real-Time Applications

iljastas, 2016-10-20:
Dear ACADO-Community,

first I want to thanks to the ACADO-Developers @ferreau @mvukov @rienq @borishouska  etc. for their great work.

**My Problem:**
I want to generate C-Code for a real-time application (vehicle control). In the first step I try to control a vehicle in a simulation environment. Fot this I'm using a really simple kinematic vehicle/robot model which is called _single track model_.  In the next step I want to use more "complex" models and try to control real vehicles/robots.

I can generate a simulation model and control it with a (N)MPC-Controller in ACADO. Everything is fine.
But if I generate a similar code and try to control an extern simulation model (in GAZEBO), it doesn't work. Obviously, I'm doing something wrong because I don't get good control vectors.
I'm using ACADO with C++, so no MATLAB or SIMULINK or any other interface.

**C++ Code for C-Code Generation (snippet of my Program):**

    // SYSTEM VARIABLES
    //
    l_vehicle = 2.78;
    v_vehicle = 0.1;
    Ts = 0.1;
    horizon_t = 5.0;

    // Model equations
    //
    f << dot(x) == v_vehicle * cos(theta);
    f << dot(y) == v_vehicle * sin(theta);
    f << dot(theta) == v_vehicle / l_vehicle * tan(phi_L);

    // Reference functions
    //
    h << x << y << theta << phi_L;
    hN << x << y<< theta;

    
    // Weighting matrices
    //
    DMatrix W = eye <double>(h.getDim());
    W(0, 0) = weight_x;
    W(1, 1) = weight_y;
    W(2, 2) = weight_theta;
    W(3, 3) = weight_phi_L;

    DMatrix WN = eye <double>(hN.getDim());
    WN(0, 0) = terminal_weight_x;
    WN(1, 1) = terminal_weight_y;
    WN(2, 2) = terminal_weight_theta;

    // MPC PROBLEM FORMULATION
    //
    int horizon_steps = horizon_t / Ts;
    ROS_INFO("horizopn_steps N = %d", horizon_steps);
    OCP ocp(0.0, 5.0, horizon_steps);

    ocp.subjectTo(f);
    ocp.minimizeLSQ(W, h);
    ocp.minimizeLSQEndTerm(WN, hN);
    ocp.subjectTo(-0.5 <= phi_L <= 1.5);

    // Export as  C-Code
    //
    OCPexport mpc(ocp);
    int Ni = 1;

    mpc.set(HESSIAN_APPROXIMATION, GAUSS_NEWTON);
    mpc.set(DISCRETIZATION_TYPE, MULTIPLE_SHOOTING);
    mpc.set(INTEGRATOR_TYPE, INT_RK4);
    mpc.set(NUM_INTEGRATOR_STEPS, horizon_steps * Ni);
    mpc.set(SPARSE_QP_SOLUTION, CONDENSING);
    mpc.set(QP_SOLVER, QP_QPOASES);
    mpc.set(HOTSTART_QP, YES);
    mpc.set(GENERATE_TEST_FILE, YES);
    mpc.set(GENERATE_MAKE_FILE, NO);
    mpc.set(GENERATE_MATLAB_INTERFACE, NO);
    mpc.set(GENERATE_SIMULINK_INTERFACE, NO);

    // Further Export Settings
    std::string path=ros::package::getPath("mpc_pathcontrol");
    std::string path_dir = path + ros::this_node::getName();

    if (mpc.exportCode(path_dir) != SUCCESSFUL_RETURN)
    {
	exit( EXIT_FAILURE);
    }

The Code is running and I get all files. Then I implemented is after the test.c

**Control the Vehicle via the C-Code (snippet of my program):**
 void MPPTC::pathcontrol() {

    static bool is_control_init = false;

    if (is_control_init == false)
    {
	// Initialize the solver.
	acado_initializeSolver();

	// Initialize the states and controls.
	for (int i = 0; i < NX * (N + 1); ++i)
	    acadoVariables.x[i] = 0.0;

	for (int i = 0; i < NU * N; ++i)
	    acadoVariables.u[i] = 0.0;

	for (int i = 0; i < NY * N; ++i)
	    acadoVariables.y[i] = 0.0;

	for (int i = 0; i < NYN; ++i)
	    acadoVariables.yN[i] = 0.0;

	// Prepare first step
	acado_preparationStep();
	is_control_init = true;
    }

    // Initialize the measurements/reference.
    // --------------------------------------
    std::vector <geometry_msgs::Pose2D> ref_path;
    get_ref_path(ref_path);

    for (int i = 0; i < ACADO_NY * ACADO_N; ++i) // NY * N = 4 * 50 = 200
    {
	int state = i % ACADO_NY;
	int index = i / ACADO_NY;

	if (state == 0) acadoVariables.y[i] = ref_path[index].x;
	else if (state == 1) acadoVariables.y[i] = ref_path[index].y;
	else if (state == 2) acadoVariables.y[i] = ref_path[index].theta;
	else if (state == 3) acadoVariables.y[i] = 0.0;
    }

    // Set the Terminal Reference
    for (int i = 0; i < ACADO_NYN; ++i)
    {
	if (i == 0) acadoVariables.yN[i] = ref_path[ref_path.size() - 1].x;
	else if (i == 1) acadoVariables.yN[i] = ref_path[ref_path.size() - 1].y;
	else if (i == 2) acadoVariables.yN[i] = ref_path[ref_path.size() - 1].theta;
    }

    // MPC: set the current state feedback
    // --------------------------------
    acadoVariables.x0[0] = pose_tf.getOrigin().getX();
    acadoVariables.x0[1] = pose_tf.getOrigin().getY();
    acadoVariables.x0[2] = tf::getYaw(pose_tf.getRotation());

    // EXECUTE THE A MPC-STEP
    acado_feedbackStep();
    acado_preparationStep();

    // APPLY the new control immediately to the process, first NU components.
    comm::DriveStamped drive_cmd;
    drive_cmd.header.frame_id = base_frame;
    drive_cmd.header.stamp = ros::Time::now();
    drive_cmd.control.steering_angle = acadoVariables.u[0];
    drive_cmd.control.speed = v_robot;
    joy_manual_pub.publish(drive_cmd); // APPLY TO THE SIMULATION / REAL VEHICLE-ROBOT

 }

iljastas, 2016-10-21:
sorry for the post. It was my stupidity: you have to apply the controller-output immediately to the process/system and then you have to call acado_preparationStep(). Like in the example...

dbworth, 2017-01-31:
@iljastas  Can this thread be closed?

closed 2017-04-18 by iljastas