acado/acado #202: C-Code Generation: linking error acado_solve()

iljastas, 2016-10-21:
If you use the auto-generated C-Code in/with C++ for fast (N)MPC, then you get maybe the compiler error: undefined reference to acado_solve(). The solution is to write **extern "C" int acado_solve( void );** before the main of your program. Like:

extern "C" int acado_solve( void );
int main() {
}

mvukov, 2016-10-21:
AFAIR `acado_common.h` handles this properly.

iljastas, 2016-10-21:
not in my case which was really strange, so i used this solution:
http://stackoverflow.com/questions/29652966/undefined-reference-to-c-functions-in-c-code

mvukov, 2016-10-21:
cf. https://github.com/acado/acado/blob/master/acado/code_generation/templates/acado_common_header.h.in

mvukov, 2016-10-21:
i.e. that's a template that gets filed in during code generation process.

DavidShawky, 2020-08-12:
I have somewhat relevant problem when compiling the project , I get undefined reference to `acadoVariables'
 more undefined references to `acadoVariables' follow
/home/david/catkin_ws/devel/lib/libnonlinear_mpc_lib.so: undefined reference to `acadoWorkspace'