acado/acado #203: qpOASES Error: Maximum number of working set recalculations performed

iljastas, 2016-10-21:
Hi guys,
it get sometimes this error from qpOASES:
ERROR:  Maximum number of working set recalculations performed

Do I have any options to control the maximum number?

Thanks a lot

rienq, 2016-10-22:
Hey,

In case you do not set this maximum number yourself, this is by default set to a very high number depending on the dimensions of you problem. So the fact that you get this error is a bad sign. However, you can always try increasing it using this option:

ocp.set( MAX_NUM_QP_ITERATIONS, ... );

Best,
Rien

closed 2016-10-22 by rienq

lpaparusso, 2019-07-08:
Hi,

I am experimenting the same problem, solving a computationally demanding MPC problem (40 independent variables). When I reach the third iteration, the algorithm fails, reporting:

ERROR: Maximum number of working set recalculations performed

Although I have increased the number nWSR to 10^7, it still gives the same error. @iljastas did you simply increase that number at that time?

@rienq when you say "the fact that you get this error is a bad sign", you mean that an incorrectly written problem could lead to this issue, right? I am sure that my problem is well defined as I have already tested it in Matlab. 

Best,
Luca

iljastas, 2019-07-09:
@lpaparusso I had a problem not because of a lot of variables, I needed a really high prediction horizon and sometimes the solution was not found or something like this...some time has passed since that problem

lpaparusso, 2019-07-09:
Hi,

thanks for your reply. Let me clarify one point: when I say "high number of variables" I mean that, even if my control inputs are two, the number of intervals in the predictive horizon is 20, and so the total number of independent variables is 20*2=40. 

Given this, I suppose my problem is the same you were experimenting, unless your issue was related to MPC feasibility.

andrescodas, 2020-05-10:
Hi.  I reported a similar issue with a simple example here:

https://github.com/coin-or/qpOASES/issues/102#issue-615438957

did you find a solution