acado/acado #204: Ccode Generation: Changing Parameter Online

iljastas, 2016-11-01:
Hey guys,

is it possible to change to change the parameter in the weighting matrices online?
I read something about ExportVariables but this class is not implemented anymore?!

Thanks for your help

rienq, 2016-11-02:
Hi!

You can have a look at this example:
https://github.com/acado/acado/blob/master/interfaces/matlab/examples/code_generation/nmpc/crane/crane.m
(it works the same from C++ by the way) The idea is to define the weighting matrix as a "BMatrix" (or boolean matrix) which defines its sparsity instead of its values, which you would typically define using a "DMatrix" (or double matrix).

I hope this helps.

Cheers,
Rien

closed 2016-11-02 by rienq

iljastas, 2016-11-03:
Thanks! Really easy:
BMatrix W = eye <BooleanType>(h.getDim());
BMatrix WN = eye <BooleanType>(hN.getDim());

And then there is the ACADOvariables in the CCode:
real_t W[ 16 ];
real_t WN[ 9 ];