acado/acado #206: CCode Generation: Changing Constraints Online

iljastas, 2016-11-03:
Hey guys,

is it possible also to change the contraints online with the ccode.
To use BooleanType or BMatrix with tempalte as BooleanType doesn't work.
Is there a possibility or a trick?

Thanks a lot!

rienq, 2016-11-04:
You can do this by setting the option 'CG_HARDCODE_CONSTRAINT_VALUES' to 'NO'..
e.g., see https://github.com/acado/acado/blob/master/interfaces/matlab/examples/code_generation/nmpc/robot/generate_code.m

This creates a variable lbValues and ubValues for the simple bound values, which you can change for each SQP iteration. In case of condensing+qpOASES, these only contain the control bounds. State bounds namely get condensed such that they become affine constraints in each QP solution. These state bounds, as well as all other (affine or possibly nonlinear) constraint values can be changed in the variables lbAValues and ubAValues.

closed 2016-11-04 by rienq

FelixChen2023, 2025-05-28:
Hello Rieng , I had same issue about adjustable constraints with NMPC, while I used OnlineData for constraint as below: 

OnlineData delta_limit omega_limit;
delta_limit = 0.2;
omega_limit = 0.3;

% 
ocp.subjectTo(- delta_limit <= delta <= delta_limit);
ocp.subjectTo(-omega_limit <= omega <= omega_limit);

% 
mpc.addOption('CG_HARDCODE_CONSTRAINT_VALUES', 'YES');

but code generation  failed with info : Requested function is not yet implemented ~

acado version : stable