acado/acado #207: QP needs to be relaxed due to infeasibility and QP solution failed

ramondalmau, 2016-11-05:
Dear all, 

I tried to run my first ACADO OCP. I formulated the following OCP:

A straight descent of an an aircraft from the state (x = 0.0, h= 36000 ft, v= 460 kt) to the state (x = 150.0 NM, h= 2000 ft, v= 140 kt). x is distance, h altitude and v speed. The goal is to minimise the flight time. The controls are Thrust and flight path angle.

This problem is very simple and works in both CasADi and GAMS using IPOPT as NLP solver and direct collocation. I also determined the box constraints of all variables with realistic values and giving enough margin to avoid infeasible solutions.

Although the simplicity of the problem, I get the following results when executing the problem:

[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/dalmau/Sofware/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

sqp it | qp its |       kkt tol |       obj val |     merit val |      ls param | 
     1 |    244 |  1.196724e+20 |  9.999995e+04 |  1.436069e+20 |  1.000000e+00 | 
[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/dalmau/Sofware/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

     2 |    161 |  8.804638e+29 |  9.999995e+04 |  1.056557e+30 |  5.000000e-01 | 
[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/dalmau/Sofware/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

     3 |    261 |  1.892390e+42 |  9.999995e+04 |  2.270868e+42 |  1.000000e+00 | 
[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/dalmau/Sofware/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

[ACADO] Error: QP solution failed
  Code: (RET_QP_SOLUTION_FAILED) 
  File: /Users/dalmau/Sofware/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 499

...

Do you have any clues of what is happening? I was searching in Google but (unfortunately) I was not able to find any help. 

My first guess is that perhaps the solver needs a better initial point or "guess" to start the SQP iterations. 

If you want to look at the code, the link is: https://bleriot.upc.edu:1215/sharing/RJvP6bxlA
password: acado

thank you in advance!
kind regards,

Ramon Dalmau

iljastas, 2016-11-07:
Hi Ramon,

did you try to loose the constraints? 
One other point is:
You wrote in your Code: OCP ocp( 0.0, T, 20);
But you have to write: OCP ocp( t_start, t_end, 20 ); whereby you have do define t_start and t_end.
I think that the algorithm try to find the the optimal T in the range of t_start and t_end. Or?

Best regards
Ilja

ramondalmau, 2016-11-08:
Dear @iljastas,
Yes, I tried to loose the constraints, but result does not improve :(

Regarding your suggestion about the start and end times of the optimisaiton, I just followed the "free end time" example of acado called "simple_ocp", which is located in the examples/getting_started folder. See:

DifferentialEquation     f( 0.0, T );     // the differential equation

//  -------------------------------------
    OCP ocp( 0.0, T );                        // time horizon of the OCP: [0,T]
    ocp.minimizeMayerTerm( T );               // the time T should be optimized

iljastas, 2016-11-08:
Hi @ramondalmau,

the example have also this line:
DifferentialEquation     f( 0.0, T );     // the differential equation

Have you tried this?

Best regards
Ilja

ramondalmau, 2016-11-08:
You are right! I completely missed that line.
However, I just tried to run again the OCP adding: 
DifferentialEquation f( 0.0, T ); // the differential equation

**and the result is identical** :( :

[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /home/dalmau/Linux_BBDD/software/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

[ACADO] Error: QP solution failed
  Code: (RET_QP_SOLUTION_FAILED) 
  File: /home/dalmau/Linux_BBDD/software/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 499

kind regards,

Ramon

iljastas, 2016-11-08:
And using as few as possible constraints lead to the same result?

ramondalmau, 2016-11-08:
Dear @iljastas 

I removed almost all constraints, just keeping state and control constraints:
    ocp.subjectTo(  140.0*kt <= v <= 500.0*kt   );     
    ocp.subjectTo( 2000.0*ft <= h <=  36000.0*ft  );     
    ocp.subjectTo(     0.0   <= x <=  150.0*nm  );     
    ocp.subjectTo( -7.0*deg <= gamma <=  0.0   );    
    ocp.subjectTo(   -4000.0 <= Th    <= 10000.0   );     
    ocp.subjectTo(       500.0 <= T   <= 2000.0   );   

**and at least now it gives a solution!** A lot of warnings and erros are given:
[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /home/dalmau/Linux_BBDD/software/acado/acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

[ACADO] Error: The integration routine stopped as the required accuracy can not be obtained
  Code: (RET_UNSUCCESSFUL_RETURN_FROM_INTEGRATOR_RK45) 
  File: /home/dalmau/Linux_BBDD/software/acado/acado/integrator/integrator_runge_kutta.cpp
  Line: 1394

 and the trajectory does not look very good, **but at least I have a solution** :)
It is a good starting point. I will investigate what could be happening by modifying the algorithm options, the initial guess and the constraints. Thank you for your valuable support!

Dalmau

I will further inve

MasiParseh, 2018-05-15:
Hello Everyone,
I am having the same problem with my code, Can anyone help me?

[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

     2 |    857 |  4.669735e+24 |  1.852857e+03 |  5.603682e+24 |  1.000000e+00 | 
[ACADO] Warning: QP needs to be relaxed due to infeasibility
  Code: (RET_RELAXING_QP) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 494

[ACADO] Error: QP solution failed
  Code: (RET_QP_SOLUTION_FAILED) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 499

[ACADO] Error: Solution of banded CP failed
  Code: (RET_BANDED_CP_SOLUTION_FAILED) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/conic_solver/condensing_based_cp_solver.cpp
  Line: 233

[ACADO] Error: Step of NLP solver failed
  Code: (RET_NLP_STEP_FAILED) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/nlp_solver/scp_method.cpp
  Line: 371

[ACADO] Error: NLP solution failed
  Code: (RET_NLP_SOLUTION_FAILED) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/nlp_solver/scp_method.cpp
  Line: 282

[ACADO] Error: Problem could not be solved with given optimization algorithm
  Code: (RET_OPTALG_SOLVE_FAILED) 
  File: /Users/mparseh/Documents/Programs/acado-master/interfaces/matlab/../../acado/optimization_algorithm/optimization_algorithm.cpp
  Line: 125

No convergence acheived