acado/acado #210: Real-time implementable or executable file from NMPC via ACADO

adexcoca, 2016-11-16:
I have been working with ACADO for NMPC development for waste heat recovery system application of heavy duty diesel engine. The MIL simulation was great and it is time to test on test-rig. For this, I have to generate an executable file from the simulink model, but I couldn't pass the following error. The below error is generated while I attempt to generate an executable file from one of the demo simulink model that I found from ACADO folder (example case for pendulum control). I tried to generate for an executable file for despace (MICROAUTOBOX Gen I). The solver used is "qpOASES3". I am using Matlab 2014b,32-bit in windows system to generate "s-function" for NMPC real-time iteration. The Microautobox has c-compiler not C++.  Please go through the error code below and suggest me any solution to fix the bug.

[image]
[image]
[image]

The major issue is related to definition of  "ACAOdata, ACADOinput and ACADOoutput." 

Do you guys have any guiding material or example case, how to generate an executable file from NMPC via ACADO that has been developed in simulink?

Second question:
Does NMPC via ACADO can be implemented through INTECRIO? My urgent need is for Microautobox, please help me to resolve the above error.

Thank you for your help in advance.

jeuonk, 2017-11-03:
Were you ever able to resolve this issue on your own? I attempting to build a model for use on a Microautobox and am running into the exact same 3 errors and 2 warnings and have yet to find a solution. Any help would be appreciated.

hcl734, 2019-08-08:
Having the same problem still. Anyone found out how to handle it?