acado/acado #214: One of the arrival cost parameters is not effective

mamaliabd, 2017-03-03:
if the Arrival cost option is enabled, then in Matlab mex file we need to define the parameter as follows
input.SAC=...
input.WL=...
input.xAC=....
where the term WL has absolutely no impact. While the others are working.
I would really appreciate it if we could see why this is not working, since as I remember in the previous versions (almost two years ago) this term has some impact.

[first_try_beam.zip](https://github.com/acado/acado/files/816936/first_try_beam.zip)

Flancis, 2017-03-13:
Hi mamaliabd,

The term WL isn't working for me either. Have you also noticed that the SAC and xAC, which represent the arrival cost term, are not changing at all before and after the simulation (the value is the same as the initialized one), which shouldn't be the case as the arrival cost term should change dynamically according to the process?

Francis

mamaliabd, 2017-03-13:
Hi Francis,

You are right, something happened with the arrival cost calculation in this current version. I assume there are some conflict with some variables in some other parts. Anyhow, I have discussed this issue with the developers and hopefully this will be solved. I was using an older version on around June 2015, and I remember that at that time the arrival cost terms were effective and helped tuning the estimator easily. If you want you can checkout the previous versions, you should be able to access to those commits on acado repo. 

Cheers,
Mohammad

Flancis, 2017-03-13:
Hi Mohammad,

Good to know. There are five branches here [https://github.com/acado/acado](url). So which one did you mean is the older one?

Also, have you considered updating the arrival cost term manually maybe using Klaman Filter?

Cheers,
Francis

mamaliabd, 2017-03-13:
I think you should checkout one of the commits either in master or stable branch.
https://github.com/acado/acado/commits/master

You can do that, and of course this could lead to some improvements in your solution, but keep in mind that still we are not sure if the implementation of the arrival cost term is included in this current version. So you can not be sure I guess.

Good luck :)
Mohammad

Flancis, 2017-03-13:
Okay, thank you.

It's true that I couldn't be sure if the arrival cost term is actually included in the current implementation even though there indeed was some improvement on the result. I'll test it on the older version then.

Cheers,
Francis

Elmanderi, 2017-10-30:
Hello, is there any update regarding arrival cost parameters in Matlab/Simulink interface?

Elmanderi, 2017-10-31:
Have you solved the problem @Flancis   ?

Elmanderi, 2017-10-31:
I have solved the problem. Here is the necessary modification in the mhe_solver_sufinction.c file;

            updateArrivalCost(0);

should be added to the places where kkt and objective value are initialized. 

            updateArrivalCost(1);

should be added to the places where kkt and objective values are updated.