acado/acado #222: getKKT undefined

EttoreObbio, 2017-06-02:
Hi,
I've used the Code generation tool and everything has worked fine, now the only thing that stops me from trying my regulator is this error in the code

Undefined identifier getKKT E0020

I've included both acado_common.h and acado_auxiliary_function.h, plus stdio.h and acado_qpoases3_interface.h but it seems that they don't contain the definition for the function.
Does anybody know where it is defined?

Thanks.

rienq, 2017-06-02:
Hi,
Are you using an older version of ACADO Toolkit? This function should namely be called "acado_getKKT" instead, unless you redefined the module name "acado" to something else but it should definitely not be called "getKKT" anymore..
Best,
Rien

EttoreObbio, 2017-06-02:
Oh, thank you a lot, this has definitively solved my issue.
I'm using the stable version on VS2017, master version have given me some issues
Thank you again

closed 2017-06-06 by EttoreObbio