acado/acado #224: Examples in the directory of ACADOtoolkit/examples/process/ do not work.

liuJwei, 2017-06-08:
Hi,
I found that all the process examples didn't work, while the examples in the directory of ACADOtoolkit/examples/getting_started/ run well. I am a beginner to ACADOtoolkit, my operating system is Ubuntu 16.04 running in the VMware Workstation. According to the outputs in the terminal (attached below), I thought the states and parameters were not transmitted to the GnuploWindow. But I don't know how to make it.

academia@ubuntu:~/ACADOtoolkit/examples/process$ ./getting_started
[ACADO] Error: Unable to plot subplot as plot data is empty
  Code: (RET_EMPTY_PLOT_DATA) 
  File: /home/academia/ACADOtoolkit/acado/user_interaction/plot_window.cpp
  Line: 841

[ACADO] Error: Unable to plot current window
  Code: (RET_PLOTTING_FAILED) 
  File: /home/academia/ACADOtoolkit/acado/bindings/acado_gnuplot/gnuplot_window.cpp
  Line: 308

[ACADO] Error: Unable to plot current window
  Code: (RET_PLOTTING_FAILED) 
  File: /home/academia/ACADOtoolkit/acado/user_interaction/plotting.cpp
  Line: 113

academia@ubuntu:~/ACADOtoolkit/examples/process$

zhangzhiming2017, 2018-09-17:
你好，请问这个问题你解决了吗？

zhangzhiming2017, 2018-09-17:
和你一模一样的问题，看源代码查问题好费事，麻烦问下你怎么解决的？

zhangzhiming2017, 2018-09-17:
> Hi,
> I found that all the process examples didn't work, while the examples in the directory of ACADOtoolkit/examples/getting_started/ run well. I am a beginner to ACADOtoolkit, my operating system is Ubuntu 16.04 running in the VMware Workstation. According to the outputs in the terminal (attached below), I thought the states and parameters were not transmitted to the GnuplotWindow. But I don't know how to make it.
> 
> academia@ubuntu:~/ACADOtoolkit/examples/process$ ./getting_started
> [ACADO] Error: Unable to plot subplot as plot data is empty
> Code: (RET_EMPTY_PLOT_DATA)
> File: /home/academia/ACADOtoolkit/acado/user_interaction/plot_window.cpp
> Line: 841
> 
> [ACADO] Error: Unable to plot current window
> Code: (RET_PLOTTING_FAILED)
> File: /home/academia/ACADOtoolkit/acado/bindings/acado_gnuplot/gnuplot_window.cpp
> Line: 308
> 
> [ACADO] Error: Unable to plot current window
> Code: (RET_PLOTTING_FAILED)
> File: /home/academia/ACADOtoolkit/acado/user_interaction/plotting.cpp
> Line: 113
> 
> academia@ubuntu:~/ACADOtoolkit/examples/process$

> Hi,
> I found that all the process examples didn't work, while the examples in the directory of ACADOtoolkit/examples/getting_started/ run well. I am a beginner to ACADOtoolkit, my operating system is Ubuntu 16.04 running in the VMware Workstation. According to the outputs in the terminal (attached below), I thought the states and parameters were not transmitted to the GnuplotWindow. But I don't know how to make it.
> 
> academia@ubuntu:~/ACADOtoolkit/examples/process$ ./getting_started
> [ACADO] Error: Unable to plot subplot as plot data is empty
> Code: (RET_EMPTY_PLOT_DATA)
> File: /home/academia/ACADOtoolkit/acado/user_interaction/plot_window.cpp
> Line: 841
> 
> [ACADO] Error: Unable to plot current window
> Code: (RET_PLOTTING_FAILED)
> File: /home/academia/ACADOtoolkit/acado/bindings/acado_gnuplot/gnuplot_window.cpp
> Line: 308
> 
> [ACADO] Error: Unable to plot current window
> Code: (RET_PLOTTING_FAILED)
> File: /home/academia/ACADOtoolkit/acado/user_interaction/plotting.cpp
> Line: 113
> 
> academia@ubuntu:~/ACADOtoolkit/examples/process$

和你一模一样的问题，看源代码查问题好费事，麻烦问下你怎么解决的？

zhangzhiming2017, 2018-09-19:
已经解决，因为代码更新了，而教程没更新

yexinyu, 2021-08-06:
> 已经解决，因为代码更新了，而教程没更新

请问如何解决的？