acado/acado #234: Compilation error using make file in getting_started_export dir of code generation tool

FilipLO, 2017-09-24:
Hello,

when trying to compile the make file in the code generation directory (/getting_started_export$ make), the following error occurs. (Everything before compiled fine.)

What has to be adapted to resolve this? Many thanks in advance. 

(PS: git rev-parse head gives 5adb8cbcff5a5464706a48eaf073218ac87c9044)

g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/Bounds.o qpoases/SRC/Bounds.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/Constraints.o qpoases/SRC/Constraints.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/CyclingManager.o qpoases/SRC/CyclingManager.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/Indexlist.o qpoases/SRC/Indexlist.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/MessageHandling.o qpoases/SRC/MessageHandling.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/QProblem.o qpoases/SRC/QProblem.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/QProblemB.o qpoases/SRC/QProblemB.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/SubjectTo.o qpoases/SRC/SubjectTo.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/Utils.o qpoases/SRC/Utils.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o qpoases/SRC/EXTRAS/SolutionAnalysis.o qpoases/SRC/EXTRAS/SolutionAnalysis.cpp
g++ -O3 -finline-functions -I. -I./qpoases -I./qpoases/INCLUDE -I./qpoases/SRC   -c -o acado_qpoases_interface.o acado_qpoases_interface.cpp
acado_qpoases_interface.cpp: In function ‘const char* acado_getErrorString(int)’:
acado_qpoases_interface.cpp:69:9: error: ‘getErrorString’ is not a member of ‘MessageHandling’
  return MessageHandling::getErrorString( error );
         ^
make: *** [acado_qpoases_interface.o] Error 1

rienq, 2017-09-24:
Hi Filip,

Well spotted! This can be fixed by copying the qpoases folder from external_packages, overwriting the version in the getting_started_export folder and typing "make clean all".
Apparently, the qpoases version in the getting started export folder, which should anyway not be there to avoid code duplication, is quite old and has never been updated..

Best,
Rien

FilipLO, 2017-09-25:
Hello Rien,

Thanks foto your swift answer. I will check it.

Regards
Filip

FilipLO, 2017-09-29:
Idd this works. Tnx a lot.

mkyun2, 2023-05-30:
GOOD!