acado/acado #235: ACADO with Simulink:  C++ Code Generation SFunction. Controller Does not work

JavierPradoRomo2007, 2017-09-26:
Hi there,
I have been testing ACADO tolkit and It is a great tool, however I am working not only in the MATLAB console. I am using it in Simulink. I based my project on the example proper of ACADO as "template" (..\code_generation\nmpc\robot\init_sim.m) because there is no any place which tells you how to generate the Sfunction.cpp. Everything seems to work, but the control output from the controller is zero. It was tested in the console environment and my project works fine. I am using an Sfunction including a file .cpp. I checked feedback states and control outputs, but Control outputs show as zero and model works correctly. Everything compiles, then I do not where to fix my problem. Can someone help me!. I will attach my example and I print where the problem is possible. Thanks beforehand for helping me.

This is the project:
[ControlAcado.zip](https://github.com/acado/acado/files/1335299/ControlAcado.zip)

JavierPradoRomo2007, 2017-09-27:
Hi there, I insist in my question. In fact, I would like to say that in the ACADO Tollkit website, there is no way to download any PDF tutorial, nor forum is opened. Please, help!

Elmanderi, 2017-10-30:
Can you clarify the problem you have faced?