acado/acado #238: Arrival cost does not get updated inside S-function

mohit004, 2017-10-24:
Dear Developers,

Just recently I realized that inside the 'nmhe_solver_sfunction.c' file, there is no function call made to 'nmhe_updateArrivalCost( )' function. Most probably, this is the reason why arrival cost parameters are not getting updated over time. The arrival cost perfectly gets updated when I write my own C++ codes for NMHE problem. Am I missing something or that's the way it is supposed to be?

Thanks in advance for the answer!

Kind Regards
Mohit

Elmanderi, 2017-10-31:
I am also facing the same problem, is there any update on that @rienq ?

Thanks.

Elmanderi, 2017-10-31:
I have solved the problem. Here is the necessary modification in the mhe_solver_sufinction.c file;

            updateArrivalCost(0);

should be added to the places where kkt and objective value are initialized. 

            updateArrivalCost(1);

should be added to the places where kkt and objective values are updated.