acado/acado #256: How can i define objective function of the problem at will when using code generation tool?

Firmamentsola, 2018-06-28:
Hallo,

i'm researching motion planning of vehicle and want to use Acado Toolkit to solve my problem in MATLAB. In the maunal  for MATLAB users i've found that the user can define Mayer term or Langrange term at will, but the programm runs very slow. Then i've tried to use the code generation tool to speed up the programm. However the tool supports only for LSQ objective function. 

So, is there some methods to define the objective function at will and at the same time to tranfer it to c code to reduce the computing time? Thank you in advance for your help.

Best wishes
Tianyu Fu

zpxiee, 2018-07-30:
I am meeting the same problem. How can I define my non-convex objective function?

dangdongfang, 2018-12-21:
hello Tianyu Fu：
I'm a PHD candidate from Chongqing University China, I'm also doing this research area. Wish to have further communication with you. My WeChat ID is    Brave-dang    and my Email: dangdongfang@cqu.edu.cn