acado/acado #261: Error when compiling: reference member 'f' binds to a temporary object whose lifetime would be shorter

Yuyang01, 2018-11-22:
acado-master/acado/code_generation/export_function_declaration.cpp:44:82: error: 
      reference member 'f' binds to a temporary object whose lifetime would be shorter
      than the lifetime of the constructed object
  ...) : ExportStatement( ), f( ExportFunction() )
                                ^~~~~~~~~~~~~~~~
/Users/Yuyang/Downloads/acado-master/acado/code_generation/export_function_declaration.hpp:104:25: note: 
      reference member declared here
                const ExportFunction& f;
                                      ^
1 error generated.
make[2]: *** [acado/CMakeFiles/acado_toolkit_s.dir/code_generation/export_function_declaration.cpp.o] Error 1
make[1]: *** [acado/CMakeFiles/acado_toolkit_s.dir/all] Error 2
make: *** [all] Error 2

Yuyang01, 2018-11-22:
Who can help me, thank you very much

tomcattigerkkk, 2019-05-08:
Do you work on OSX? @Yuyang01

dpinte, 2019-05-12:
@tomcattigerkkk got the same issue. The fix suggested by @heyyeh12 solves the problem.

Yuyang01, 2019-05-12:
> Do you work on OSX? @Yuyang01

It works on my mac.

whateverforever, 2021-02-10:
Had the same issue on macOS big sur, fix by @heyyeh12 works :)