acado/acado #263: issue regardind code generation with matlab

spdiwan, 2018-12-31:
I am trying to implement the example given in the ACADO toolbox of inverted pendulum using matlab. I have following error

Warning: The file 'E:\temp_nmpc_final\acado_codegeneration_ROTPEN\ACADO_swingup.m' could not be cleared because it contains
MATLAB code that is currently executing. 
> In ACADO_swingup (line 2) 
Warning: The file 'E:\temp_nmpc_final\acado_codegeneration_ROTPEN\ACADO_swingup.m' could not be cleared because it contains
MATLAB code that is currently executing. 
> In ACADO_swingup (line 2) 
Warning: The file 'E:\temp_nmpc_final\acado_codegeneration_ROTPEN\ACADO_swingup.m' could not be cleared because it contains
MATLAB code that is currently executing. 
> In ACADO_swingup (line 2) 
Writing c++ files... 
Compiling c++ files... 
Building with 'Microsoft Visual C++ 2017'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 
[ACADO] Information: Code generation successful
  

compiling... Building with 'Microsoft Visual C++ 2017 (C)'.
MEX completed successfully.
done! --> ../integrate_pendulum.mexw64
Writing c++ files... 
Compiling c++ files... 
Building with 'Microsoft Visual C++ 2017'.
MEX completed successfully.

ACADO Toolkit for Matlab - Developed by David Ariens and Rien Quirynen, 2009-2013 
Support available at http://www.acadotoolkit.org/matlab 
 

ACADO Toolkit::Code Generation Tool
Copyright (C) 2008-2014 by Boris Houska, Hans Joachim Ferreau,
Milan Vukov, Rien Quirynen, KU Leuven.
Developed within the Optimization in Engineering Center (OPTEC)
under supervision of Moritz Diehl. All rights reserved.

ACADO Toolkit is distributed under the terms of the GNU Lesser
General Public License 3 in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Lesser General Public License for more details.

compiling... Building with 'Microsoft Visual C++ 2017'.
MEX completed successfully.
done! --> ../acado_MPCstep.mexw64
------------------------------------------------------------------
               Simulation Loop
------------------------------------------------------------------
Error using acado_MPCstep
Field yN must be of size: 1 x 4.

Error in ACADO_swingup (line 173)
    output = acado_MPCstep(input);
 
>>

rienq, 2019-01-04:
You should read and follow the error message
`Field yN must be of size: 1 x 4.`

closed 2019-01-04 by rienq

spdiwan, 2019-01-04:
i have gone through the code but its default example code came with acado toolbox for inverted pendulum.
It has been verified by ACADO tool developer, I think so.....