acado/acado #274: Making of MATLAB interface fails  MacOS Catalina

ragiadel, 2019-11-29:
When I try to use make command in MATLAB 2018a it fails after 30% completion I tried so many tricks I found  but none worked. Here is the error i get if anyone can help me please : 

Error using mex
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function_declaration.cpp:34:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function_declaration.hpp:39:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function.hpp:39:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_variable.hpp:35:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_argument.hpp:38:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/matrix_vector/matrix_vector.hpp:37:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/matrix_vector/vector.hpp:40:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/matrix_vector/matrix_vector_tools.hpp:39:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/Dense:1:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/Core:279:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Core/MatrixBase.h:138:
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Core/../plugins/MatrixCwiseUnaryOps.h:62:32:
warning: 'binder1st<std::__1::equal_to<float> >' is deprecated [-Wdeprecated-declarations]
inline const CwiseUnaryOp<std::binder1st<std::equal_to<Scalar> >, const Derived>
                               ^
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Core/PlainObjectBase.h:85:32:
note: in instantiation of template class 'Eigen::MatrixBase<Eigen::Matrix<float, 4, 1, 0, 4, 1> >'
requested here
class PlainObjectBase : public internal::dense_xpr_base<Derived>::type
                               ^
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Core/Matrix.h:128:12:
note: in instantiation of template class 'Eigen::PlainObjectBase<Eigen::Matrix<float, 4, 1, 0, 4, 1> >'
requested here
  : public PlainObjectBase<Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols> >
           ^
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Geometry/Quaternion.h:284:16:
note: in instantiation of template class 'Eigen::Matrix<float, 4, 1, 0, 4, 1>' requested here
  Coefficients m_coeffs;
               ^
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../external_packages/eigen3/Eigen/src/Geometry/arch/Geometry_SSE.h:21:35:
note: in instantiation of template class 'Eigen::Quaternion<float, 0>' requested here
  static inline Quaternion<float> run(const QuaternionBase<Derived>& _a, const
  QuaternionBase<OtherDerived>& _b)
                                  ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1033:28:
note: 'binder1st<std::__1::equal_to<float> >' has been explicitly marked deprecated here
class _LIBCPP_TEMPLATE_VIS _LIBCPP_DEPRECATED_IN_CXX11 binder1st
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1101:39:
note: expanded from macro '_LIBCPP_DEPRECATED_IN_CXX11'
#  define _LIBCPP_DEPRECATED_IN_CXX11 _LIBCPP_DEPRECATED
                                      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1090:48:
note: expanded from macro '_LIBCPP_DEPRECATED'
#    define _LIBCPP_DEPRECATED __attribute__ ((deprecated))
                                               ^
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function_declaration.cpp:34:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function_declaration.hpp:39:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_function.hpp:39:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_variable.hpp:35:
In file included from
/Users/ragiadel/ACADOtoolkit/interfaces/matlab/../../acado/code_generation/export_argument.hpp:38:
[... 323 lines cut ...]
-1> >::operator+=<Eigen::Matrix<double, -1, -1, 1, -1, -1> >' requested here
        elements[rowIdx][colIdx] += tmp;
                                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1033:28:
note: 'binder1st<std::__1::equal_to<double> >' has been explicitly marked deprecated here
class _LIBCPP_TEMPLATE_VIS _LIBCPP_DEPRECATED_IN_CXX11 binder1st
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1101:39:
note: expanded from macro '_LIBCPP_DEPRECATED_IN_CXX11'
#  define _LIBCPP_DEPRECATED_IN_CXX11 _LIBCPP_DEPRECATED
                                      ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:1090:48:
note: expanded from macro '_LIBCPP_DEPRECATED'
#    define _LIBCPP_DEPRECATED __attribute__ ((deprecated))
                                               ^
7 warnings and 1 error generated.

Error in makehelper>execute_command (line 338)
    eval(s);

Error in makehelper (line 243)
                    execute_command (cmd, DEBUG, SRC{i}, ~PARALLEL) ;

Error in make (line 79)
    makehelper(0, {}, varargin);

Raphtor, 2020-11-10:
I too have this issue.

Anand-Padmanabhan, 2021-11-10:
I'm having the same issue. Are you using Xcode Clang++?