acado/acado #279: Error using mex

lalitkishore99, 2020-04-22:
**Getting the following error during ACADO installation:**

make clean all;
Cleaning up all ACADO files... 
Removing ACADO folders from Matlab path... 
Clean completed. 
Making ACADO... 
*Building with 'MinGW64 Compiler (C++)'.
MEX completed successfully.
*Building with 'MinGW64 Compiler (C++)'.
Error using mex
In file included from C:/Users/lalit/Dropbox/Lalit Research/Lalit/Models/acado-master/acado-master/acado/utils/acado_utils.hpp:39:0,
                 from C:\Users\lalit\Dropbox\Lalit Research\Lalit\Models\acado-master\acado-master\acado\utils\acado_message_handling.cpp:37:
C:/ProgramData/MATLAB/SupportPackages/R2019a/3P.instrset/mingw_w64.instrset/lib/gcc/x86_64-w64-mingw32/6.3.0/include/c++/cmath:1225:11: error: '::round' has not been
declared
   using ::round;
           ^~~~~

Error in makehelper>execute_command (line 338)
    eval(s);

Error in makehelper (line 243)
                    execute_command (cmd, DEBUG, SRC{i}, ~PARALLEL) ;

Error in make (line 79)
    makehelper(0, {}, varargin);

Mosolim, 2020-05-05:
i have the same problem

Mosolim, 2020-05-05:
You have to change the compiler to Windows Visual 2019 (i am using WIn 10 and Matlab 2020a) 
 make clean all;
Cleaning up all ACADO files... 
Removing ACADO folders from Matlab path... 
Clean completed. 
Making ACADO... 
*Building with 'Microsoft Visual C++ 2019'.
MEX completed successfully.

mathosborne, 2020-06-30:
I had the same problem,
To change C++ Compiler enter: 
>> mex -setup
then click on the link to Microsoft Visual C++ 2019 for the Compiler
then do the same for the link to the C++ compiler language and change it from 'MinGW64 Compiler (C++)'
 to 'Microsoft Visual C++ 2019'  and then it works ok.

closed 2020-06-30 by lalitkishore99