acado/acado #291: Installation in Mac OS

RodrigoCodeBernardo, 2021-11-24:
I have this error, when I try to run the command:

sudo port install gcc47 cmake git-core gnuplot doxygen graphviz

```
--->  Fetching archive for cloog
--->  Attempting to fetch cloog-0.18.4_4.darwin_20.x86_64.tbz2 from https://packages.macports.org/cloog
--->  Attempting to fetch cloog-0.18.4_4.darwin_20.x86_64.tbz2 from https://fra.de.packages.macports.org/cloog
--->  Attempting to fetch cloog-0.18.4_4.darwin_20.x86_64.tbz2 from https://nue.de.packages.macports.org/cloog
--->  Building cloog
Error: Failed to build cloog: command execution failed   
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_cloog/cloog/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets if you believe there
is a bug.
```

Has it happened to anyone? How can I solve it?

Yersaiyn-JAVA, 2022-04-16:
try to download the newer version of gcc compiler. For example, I used command: "sudo port install gcc11"

Yersaiyn-JAVA, 2022-04-16:
also try to download each port individually. For ex. first: "sudo port install cmake", then "sudo port install gnuplot", and so on ...